probonopd/MiniDexed #583: Changing performances with rotary encoder crashes in USB Gadget Mode

probonopd, 2023-12-06:
On a Raspberry Pi Zero 2 with a 0.96" OLED via i2c and no i2s DAC:

* Installed [MiniDexed_2023-12-05-e5b2656](https://github.com/probonopd/MiniDexed/suites/18779064606/artifacts/1094269937)
* Enabled USB Gadget Mode
* Set sound output to HDMI
* Attached Pi to host computer (in this case, a Raspberry Pi 5) via the original Raspberry Pi USB Keyboard and Hub
* Using REAPER, played a MIDI song out to the MiniDexed USB Gadget port. Works beautifully
* Using the rotary encoder, loaded performance "Concert D" while the MIDI was playing. No issues so far
* Same for "The Grand", "Rock Piano", "Cinematic", "Piano + Pad", "SimpleM.CP-70", "Phil's CP-70", "mDX-Ballade". No issues so far
* When I select "Legend'83" it crashes and I hear a looping sound. 

I can reproduce this. It is the 9th time I try to load a performance that I get the crash.

When the crash happens, `dmesg` on the host PC says

```
sound midiC0D0: rawmidi drain error (avail = 3353, buffer_size = 4096)
```

(Note: This does not happen when I don't use USB Gadget Mode but attach a MIDI controller instead. Then I can load performances more often than 8 times both with the rotary encoder and via MIDI without the crash.)

Additional information:

* The freeze only happens when the device is connected to a host PC in USB gadget mode
* The freeze also happens when I am not playing a MIDI at the same time
* The freeze does not happen when USB gadget mode is disabled (even if powered via the same USB port at the same host PC)
* The freeze does not always happen exactly at the 9th performance, it seems to be more like a ~1/10th probability of a freeze on every performance change

diyelectromusic, 2023-12-06:
Some additional thoughts:
- Is it a genuine freeze of MiniDexed or just the PC/USB locking up?  i.e. do the menus still work?  Can serial MIDI work? Does sound all lock up?
- Can you try different RPi versions in Gadget mode too?  What else do you have access to?
- Can you use a different host computer to attempt to reproduce the problem?
- If you use a different MIDI file do you get the same issue?
- If you enable MIDI debugging what is the last thing on the display when the problem occurs - is it always the same?  Assuming MIDI debugging doesn't slow everything down so much it all starts working of course :)
- Can you route the MIDI through a monitor application on the PC to see what is being sent at the PC end when it fails?
- What MIDI channels are being used in the MIDI file and what MIDI channel is being used for the system PCCH channel?
- If in USB gadget mode, but MIDI is sent over serial, does it still cause an issue?
- Can we eliminate the other hardware somehow (not sure how - try a different screen?  I'm guessing it isn't likely to be useful to try to reproduce the fault when running headless, e.g. selecting performances over MIDI...
- In fact, try to reproduce the problem when changing performances via MIDI - e.g. using some kind of MIDI merge application or hardware, or maybe USB gadget for the MIDI stream and serial MIDI for the PCCH messages.

Given the answers to the above, I expect we'll need to produce some kind of debug version with extra printing to try to work out what is going on.  You might have to send me your MIDI file too - although the chances of reproducing it might be a bit slim, I'll certainly have a go!  If it turns out it is only one specific MIDI file anyway, then email that over and I'll have a look at it anyway.

The key in all this, as I'm sure you know, is changing one thing at a time and documenting all the results :)

As gadget mode is so new, it could even be an issue with the USB drivers in circle itself...

Kevin

diyelectromusic, 2023-12-06:
Actually can you check your 'MIDIAutoVoiceDumpOnPC' setting?  If this is enabled (I left this as the default on the assumption someone decided it should be enabled in the system when I was adding the option!) then every voice change triggers a sysex dump to all connected MIDI ports.  Selecting a performance will trigger 8 individual voice sysex dumps potentially back over the USB Gadget link...

If that is enabled, try disabling it and see if that makes a difference.

Kevin

Banana71, 2023-12-06:
On a Raspberry Pi Zero 2 with a HD44780 LCD via i2c and with i2s DAC GY PCM5102:

Installed [MiniDexed_2023-12-05-e5b2656](https://github.com/probonopd/MiniDexed/suites/18779064606/artifacts/1094269937)
Enabled USB Gadget Mode
Set sound output to i2s
Attached Pi to host computer (PC Windows 11)

No crashes observed so far, all performances play perfectly. But after selecting a performance with the encoder or via MIDI, the display remains frozen for 3 seconds. The rotations on the encoder are recorded and after three seconds the correct performance is displayed according to the rotation. The sound output works during this time.

diyelectromusic, 2023-12-06:
I've just sat here randomly selecting performances using an encoder on a Pi 3A+ in gadget mode and have seen no freezing or pausing so far.  I've also confirmed that yes, I do get a whole pile of sysex dumps back over the USB link for every performance change!

Kevin

Banana71, 2023-12-06:
`MIDIAutoVoiceDumpOnPC` has no influence on the behavior for me

probonopd, 2023-12-08:
At the moment I'd say this is too obsure an issue to spend much time on right now. I can reproduce it, but it's not really a problem for me. So I'd say let's leave this issue open as a reminder in case others experience it, too, but let's not spend time on it at the moment.

diyelectromusic, 2024-01-31:
Can you see if this is still an issue with the latest build?

If so, we need a means of reproducing it that can be duplicated by someone else and ideally answers to the questions I posed above.

Kevin