probonopd/MiniDexed #587: Rename "MIDI Garget MIDI 1" to "MiniDexed MIDI 1"

probonopd, 2023-12-09:
Currently, REAPER (on WINE) shows MiniDexed as "MIDI Gadget MIDI 1".

We should rename "MIDI Gadget MIDI 1" to "MiniDexed MIDI 1" if possible.

probonopd, 2023-12-09:
Is it coming from Circle here?

https://github.com/rsta2/circle/blob/c21f2efdad86c1062f255fbf891135a2a356713e/lib/usb/gadget/usbmidigadget.cpp#L166

Banana71, 2023-12-09:
The idea is very good. My other MIDI devices are also displayed more clearly.
Other manufacturers do not agree whether the name "MIDI" should still be in the name. But "MiniDexed MIDI" is also great.

[image]

probonopd, 2023-12-09:
I am on it.

probonopd, 2023-12-09:
Build for testing:
[MiniDexed_2023-12-09-a1518d6](https://github.com/probonopd/MiniDexed/suites/18901668257/artifacts/1103715663)

probonopd, 2023-12-09:
Also, we should probably disable the MIDI output as it is currently unused.

diyelectromusic, 2023-12-09:
No, I really disagree with patching code like this.  If we want it changing we need to log an issue in circle to make it configurable. This kind of build-time patching will just make things very hard to maintain - and let's face it, with quite a few people pitching in changes over the evolution of MiniDexed, it's already quite hard to maintain the codebase... (hence my cautious approach with the performance subdirectories)

Kevin

probonopd, 2023-12-09:
I have opened https://github.com/rsta2/circle/issues/416.

probonopd, 2023-12-10:
@diyelectromusic Do you think you could change this PR to follow https://github.com/rsta2/circle/issues/416#issuecomment-1848919646?

diyelectromusic, 2023-12-10:
I'll have a look at some point. I'm still chewing over performance things at the moment.

Kevin

closed 2023-12-30 by probonopd