probonopd/MiniDexed #595: Performances Menu Not Showing Up

RK11111111111, 2024-01-01:
I can't seem to get the performance menu or effects menu to show up.  I am using the 4 key midi navigation method.  I downloaded last nights 12/31/2023 build.  I downloaded the zip, copied over the all the files added the sysex directory.  I enabled the sound hat i2s and the lcd i2c.  Then set the midi channel to 1 and mapped the four navigation buttons.

probonopd, 2024-01-01:
Hi @RK11111111111 
Which Raspberry Pi model are you on?
Can you navigate through the other menus?

RK11111111111, 2024-01-01:
I was able to figure it out, it was at the end of all the other TG, but now when I select a performance there isn't any sound.

RK11111111111, 2024-01-01:
If I edit the performance.ini, and store it as the second performance it loads.

RK11111111111, 2024-01-01:
Weird, but then if I load the default and reload my new performance, the midi channel the keyboard is on changes.

RK11111111111, 2024-01-02:
I think I figured it out skanky sysex files.

RK11111111111, 2024-01-02:
Actually, no something else.  It seems that any time I use the select or prev key, it is changing the TG that is mapped to midi channel 1.  With the defualt performance they are the same voice or are inactive, so I didn't notice it.

So if I have eight TG one through eight each with a different voice, as I navigate the menu pressing select it changes the keyboard cycles through the TG, when I hit previous it cycles the other way, both changing the menu and the TG.  Oddly enough, it somehow only effects the keyboard, though I have one of the drum pads also mapped to midi channel one, and it doesn't get changed, I'm not sure how that would even be possible.   Furthermore, I've tried mapping the midi buttons to different CC ranges 20-23,100-103,110-113, and they all seem to be doing the same thing.  To be more specific, when I load a performance, it loads and all of the channels except 1 seem to be affected.  I also tried mapping CC buttons to a different channel. 

My performance.ini ideally has 8 TG each mapped to a different midi channel.  Do I have a setting off somewhere, I know there is one mode where you can cycle through the voices, but I can't seem to figure out how to turn that off.

probonopd, 2024-01-02:
Having a hard time understanding what exactly you are doing. Can you please share more information:
* The hardware you are using
* Your `minidexed.ini` file
* Which buttons you are pressing in which order on which device(s)
* What you are seeing at each point on the display
* What happens/what you hear vs. what you would expect to happen/hear

RK11111111111, 2024-01-02:
[example.txt](https://github.com/probonopd/MiniDexed/files/13813898/example.txt)

I am attaching an example.txt file, which is a performance.ini.

In addition on my raspberry pi, I have the default DX7 soundbank, I downloaded from the repo eg MiniDexed-sysexFinal 0_Original_Yamaha, 0_DX7.  

When, I load the performance, and I play my keyboard on midi channel 1, TG1 doesn't play, but some other TG from the file.  In addition when I try to navigate with the prev and select keys, the sounds that the keyboard produces changes.  Each time I go up or down a level in the menu system.  As if they are changing the TG at the same time as navigating the menu.  I can provide the minidexed.ini also.

diyelectromusic, 2024-01-02:
I've replied to your discussion question (https://github.com/probonopd/MiniDexed/discussions/596), although I'm not sure I'm totally getting what you're trying to do :)

But you might be hitting an issue related to the fact that MIDI channels per tone generator are stored in the performance files - so each file that can be selected (assuming you're using buttons/MIDI to change performances, not voices) will have to have the same per TG MIDI config or things will get very confusing pretty quickly :)

So as well as your example performance.ini can you show us one you're switching to?

Kevin

RK11111111111, 2024-01-02:
I was booting to the default then trying to switch to the new one.

I have also tried booting straight to that performance which I linked above, and the issue happens immediately.  I don't have to load the performance file, the prev and select buttons just start switching TG.  Which is kind of why I started this thread to begin with.  I didn't know what was going on, and couldn't find the performance menu.

diyelectromusic, 2024-01-03:
I think at this point we really need to see your minidexed.ini to see if there is an odd overlap of button configurations or something. I'm afraid there isn't much sanity checking on the prospect of button configuration overlaps and things (https://github.com/probonopd/MiniDexed/issues/336),

Kevin

RK11111111111, 2024-01-03:
[minidexed.txt](https://github.com/probonopd/MiniDexed/files/13819459/minidexed.txt)

This is the most recent version.  I can't gaurantee I didn't walk on something.

I recently, reflashed the eprom, reformatted the sd card, installed the most recent version of minidexed, then changed the i2s sound card setting, added the LCD address, set the midi button channel, at this point the issue started.

Further tests that I did included adding channel 9 just to be sure that it didn't have a conflict with the 1-8 channels for navigation then added the midi buttons cc110-cc113.  In addition I disabled anything else that looked remotely conflicting.  EG the rotary encoder, any of the unassigned button addresses were set to zero.  Disabled program change messages etc.

I am using Alesis V25 keyboard.

diyelectromusic, 2024-01-03:
I'm struggling to see what is going on tbh.

Note: there are errors on startup with this config about invalid actions, but I think that is because you've set ButtonActionPgmUp=0 (and so on) when a disabled action is just ButtonActionPgmUp= (i.e. equals nothing) - so I don't think that will be an issue.

It causes a warning that pin 11 is assigned several times, but again I think that is just because the default configuration has a click, doubleclick, and longpress as well as being for the shortcut action... so I don't think that is an issue...

... but grasping at straws, I'd suggest fully turning off all GPIO buttons:
```
ButtonPinPrev=0
ButtonActionPrev=
ButtonPinNext=0
ButtonActionNext=
ButtonPinBack=0
ButtonActionBack=
ButtonPinSelect=0
ButtonActionSelect=
ButtonPinHome=0
ButtonActionHome=
ButtonPinShortcut=0
```

I don't have an easy way to test your MIDI configuration, but the CC messages have to act as button presses themselves - i.e. they need an Off->On->Off transition to trigger as a press and release.

So, for example, to select will require the MIDI messages:
B8 6E 40 -> B8 = MIDI CC message on channel 9; 6E=CC controller 113; 40 = value 64 (>=64 is considered ON).
B8 6E 00 -> As above but with value 00 (<64 is considered OFF).

At this point it might be worth turning on MIDI debugging and seeing what is actually being sent from the Alesis.  Use MIDIDumpEnabled=1 in minidexed.ini

But in general I'm afraid I'm losing track as to what you're observing with which configurations and what the actual issue is - I'm not saying you don't have one, I'm saying I've lost track of what I'm attempting to remotely debug!  Sorry :)

Kevin

RK11111111111, 2024-01-03:
It's pretty simple, you can load as many TG and midi channels as you want as long as you want exactly one TG and one midi channel.

I don't have a display, how do I view the midi dumps?

diyelectromusic, 2024-01-03:
I'm sorry, but that doesn't match anything related to my experiences with MiniDexed, so I still don't really know what the issue is you're seeing I'm afraid.

You need a HDMI display for MIDI dumps and any other messages that it might be complaining about.

I think at this point I'd need to know the steps you've taken from a fresh install and what behaviours you're seeing exactly.  I don't know what you mean by "load a TG" - all tone generators are initialised and running on all versions of a Pi apart from a V1 and the V1 Zero.  In all this time, you still haven't actually said even what hardware you're using so I'm largely just guessing at the moment.

It may be that you'll need to start with a more conventional user interface configuration (which means encoder and screen) to get a feel for what the menus should be doing then we'll all have a baseline for what is differing when you control it over MIDI.  MiniDexed supports many different options which is both a blessing and a curse, but we need something to compare it all to I'm afraid.

You haven't said if you've tried my suggestions in my previous post and there are still a few unanswered questions from previous replies too.  I'm still wondering if there is a weird interaction between the MIDI buttons and the "ButtonPinShortcut" option...

Kevin

RK11111111111, 2024-01-03:
Ah, I have a raspberry pi 4 with a raspiaudio I2S hat and a QAPaAS LCD I2C

diyelectromusic, 2024-01-03:
Right... are you using the last release or the build from the most recent PR that is going on?  I've used the last release from here: https://github.com/probonopd/MiniDexed/releases and MIDI buttons seem to be fine with my Aturia.

But they don't work with that latest PR due to a bug I've just found (and am about to fix)!

Kevin

RK11111111111, 2024-01-04:
I've tried both the 12/3 and 12/31 builds.  I set the Alesis buttons to momentary, pressed =127 release =0
 I've tried channel 1 and 9 as well as CC ranges 20-23, 100-103, and 110-113

diyelectromusic, 2024-01-04:
There is no 12/3 from the releases area... was the 12/31 the official last "continuous" release build?  i.e. this one: https://github.com/probonopd/MiniDexed/actions/runs/7370558303

This is the one I tried and MIDI buttons worked as I expected with your configuration...

Kevin

RK11111111111, 2024-01-04:
MiniDexed_2023-12-03-c6325cd

RK11111111111, 2024-01-04:
Breakthrough.  If I load the performance, the keybed plays a weird channel, but if I unplug the Alesis from the usb port, then plug it back in it starts playing the right channel.

RK11111111111, 2024-01-04:
So, yeah, pretty much anything that sends midi, breaks the alesis.  Just adjusting the modulation menu, I have to plug and unplug the keyboard.

diyelectromusic, 2024-01-05:
Sorry, no idea what's going on there then...

As I say, all I can recommend at this stage is go to the last continuous release and a more traditional MiniDexed setup hardware wise and then see how changing things changes how the system behaves.

Kevin

RK11111111111, 2024-01-05:
I tried commenting out the Sysex Send in a forked repo, didn't help, but my mini-hdmi cable arrived today:
[image]

This is after I unplugged then replugged the device.  You can see the USB initialization, it seems to be treated as two separate devices, I guess for the pads and keybed.
I pressed a key
...
MIDI0: 80 30 00
....
Then hit a pad
...
MIDI0: 92 28 79
...
Then navigated the menu with midi buttons.
B8...
B8...
B8...
Loaded a performance
B8 71 ...
Now the interesting part, when I play the same key now
91 30 23
...
But the pad is the same
MIDI0: 92 28 79

I did this a couple of times plugging and unplugging the usb cable for the keyboard, and the hex that the key device switched to changed each time.

Maybe its an issue with the USB gadget mode?

diyelectromusic, 2024-01-05:
I still don't really know what's going on tbh, but on those MIDI messages:
80 30 00 is a NoteOff event on channel 1 for note 48 (C3)
91 30 23 is a NoteOn event on channel 2 for note 48 (C3) with a velocity of 35

So they relate to different things...

But from your photo, I can only see pairings of NoteOns and NoteOffs, so not sure which numbers you're looking at in your description?

e.g.
90 30 23 - 80 30 00 - Playing note 48 on channel 1
92 28 79 - 82 28 00 - Playing note 40 on channel 3
91 30 23 - 81 30 00 - Playing note 48 on channel 2
92 28 63 - 82 28 00 - Playing note 40 on channel 3

If the Alesis is plugging into the Pi, then USB gadget mode shouldn't be in use (you've not configured that I assume).  All of those MIDI messages are what MiniDexed is receiving, presumably from your Alesis...

Most USB devices have several functions. I forget exactly but many USB MIDI controllers come up with those two functions, but only one is the MIDI stream.

> As I say, all I can recommend at this stage is go to the last continuous release and a more traditional MiniDexed setup hardware wise and then see how changing things changes how the system behaves.

Kevin

RK11111111111, 2024-01-20:
I thought I had already attached the minidexed.ini, and described the sequence of presses, and shown the output, but here goes...

My minidexed.ini and performance.ini are attached.  I renamed them to txt as ini files are blocked.

At first boot the screen shows.

Minidexed TG1 

and plays the performance.ini as expected.

I press select(the midi key I have mapped to select in the mididexed.ini).  This shows the TG1 menu specifically the screen says.

Voice

However, now if I play the keybed.  I hear the second tone generator TG2.

If I press select again, I see the voice in the bank, however now the keyboard is playing TG3.

If I press the back key to go up a menu.  I see

Voice 

However, now I hear TG2 when I play the keyboard.  If I press  back again, I am back at the main menu.

Minidexed TG1,

I hear the first tone generator TG1 in the keyboard.

Now, I can go further if I am in the Voice menu and continue to press select.  I can iterate through each of the tone generators TG3,TG4,TG5, however the menu does not change from the voice.

If I am at the main menu. 

Minidexed TG1,

I can navigate using forward and backward

Forward
TG2
Keyboard continues to play TG1 
Forward
TG3  
Keyboard continues to play TG1
....
Forward
Effects
Keyboard continues to play TG1
Performances
Keyboard continues to play TG1
select
keyboard plays TG2
Load
select
keyboard plays TG3
Default[L]
forward
more_fx(my performance.ini)
select 
keyboard plays TG4 of the new performance.
prev
Load plays TG3
prev
Performances plays TG2
prev
Perormances plays TG1

Note, how I had to push prev one extra time to get back to TG1, for example if I navigate through different performances and load each one I have to hit prev once for each time I hit load to get back to TG1.

[minidexed_ini_in_text.txt](https://github.com/probonopd/MiniDexed/files/13998000/minidexed_ini_in_text.txt)
[performance_ini_copy_in_text.txt](https://github.com/probonopd/MiniDexed/files/13998001/performance_ini_copy_in_text.txt)

RK11111111111, 2024-01-20:
Also, I am using an Alesis V25 keyboard and a raspberry pi 4B, with a raspiaudio hat.

The Alesis appears as two devices one for the pads and one for they keybed.  I have the Alesis keybed in channel one and each of the pads mapped to channels 1-8.  This is flexible, I tried different combinations and channels, eg keybed on channel 9.

diyelectromusic, 2024-01-20:
Note - this is a different minidexed.ini to the one you posted before, but in this one you appear to have both PGMUp/Down mapped to the same MIDI buttons as Prev/Next...?
```
MIDIButtonPrev=100
MIDIButtonNext=101
MIDIButtonBack=102
MIDIButtonSelect=103
MIDIButtonHome=0
MIDIButtonPgmUp=101
MIDIButtonPgmDown=100
```

So a summary of my recommendations that I think are still to be tried so far:

> But you might be hitting an issue related to the fact that MIDI channels per tone generator are stored in the performance files - so each file that can be selected (assuming you're using [...]

and

> It may be that you'll need to start with a more conventional user interface configuration (which means encoder and display with debug messages on the HDMI screen) to get a feel for what the menus should be doing then we'll all have a baseline for what is differing when you control it over MIDI.

and

> As I say, all I can recommend at this stage is go to the last continuous release and a more traditional MiniDexed setup hardware wise and then see how changing things changes how the system behaves.

and

> But from your photo, I can only see pairings of NoteOns and NoteOffs, so not sure which numbers you're looking at in your description?  If the Alesis is plugging into the Pi, then USB gadget mode shouldn't be in use (you've not configured that I assume). All of those MIDI messages are what MiniDexed is receiving, presumably from your Alesis...

The biggest issue is that whilst it may be very clear to you what you're changing and how that is changing what you are experiencing, we have no real way to know what change has resulting in which comment from you, making it really hard to string together a sequence of what exactly is happening.

I really do think you need to get a typical MiniDexed up and running and then start to configure your system so you can see what changes are inducing what behaviour.

I suspect actually there are different configuration issues popping up as things are getting changed that are perhaps unrelated to each other.

By the way, the "Select and swap TG" I think is part of the "short-but" behaviour, but as I don't use it myself, I can't say for sure.  It might be worth setting ButtonPinShortcut=0.  Although I thought this was only a rotary encoder feature, so it might be nothing...

I can't see how pressing select is (if I understand you correctly) playing the next TG.  Surely that relates to which channel you are sending notes on as to which TG is being played - is your Alexis still sending on the same channel (and if so, how do you know)?

We will need to see the MIDI debug log for the above behaviour to see what is going on.

This is the biggest issue here for me in trying to help.  We keep seeing snippets of information, but to debug it remotely, we need a consistent set: your configuration, the actions, the debug log - all from the same action, not from different experiments.

Kevin

RK11111111111, 2024-01-21:
Is there an easy way to get the midi debug log, is it possible to write it to an SD card.  It didn't write it to the screen.  It looked like I needed to recompile it for debugging.  I had already enabled the midi dumps in the ini.

RK11111111111, 2024-01-21:
> e.g. 90 30 23 - 80 30 00 - Playing note 48 on channel 1 92 28 79 - 82 28 00 - Playing note 40 on channel 3 91 30 23 - 81 30 00 - Playing note 48 on channel 2 92 28 63 - 82 28 00 - Playing note 40 on [...]

It appears to be switching channels on the keybed(the device the keyboard is sending messages on).

I suspected that it was sending messages, so I tried disabling any midi sends that I could find, as there is no good reason for minidexed to be sending these, especially in a broadcast mode.  It didn't make any difference, but I did notice that minidexed is registering multiple midi devices on the network so there is an abstraction layer where minidexed is mapping usb devices to midi channels.  Which is consistent with being able to unplug the device the midi device from the usb port then plug it back in and the tone generators be mapped correctly.

RK11111111111, 2024-01-21:
Also, minidexed being a single channel single tone generator fm synth is covered here for your referece: https://github.com/probonopd/MiniDexed/discussions/596

diyelectromusic, 2024-01-21:
The MIDI dump is the only MIDI debug log, and yes, that is enabled in minidexed.ini and only goes to the screen.  But this only logs what MiniDexed receives.

If you're suggesting MiniDexed is transmitting something that is changing your Alesis, then I've no idea what that could be.  I've not heard of anything like that so far.  I think you'd need to insert something between the Alesis and the MiniDexed, which probably means messing around with MIDI THRU and something like Hairless MIDI to pipe everything through a tool like MidiOx to see what is actually going on.

Once again though, I'm afraid I'm not keeping up with what you are reporting against what changes and experiments you've tried.

So to reiterate my previous request, please enable the MIDI dump and show us the behaviour you've described in your previous post about menu changes causing different TGs to play. I think we need to focus on a single issue at a time.

I cannot stress enough the importance of a known starting configuration and a consistent set of actions, the consequences, and the MIDI dump of those actions, in order for us to be able to move this forward.  Without this, I'm afraid I'm going to have to stop spending time thinking about this.

For example, I don't know what this is in reference to and why it is relevant to the issue or it's diagnosis;
> Also, minidexed being a single channel single tone generator fm synth is covered here for your referece

Once again apologies, if I'm missing something, but I'm not going to be able to help any more without a coherent set of information I can make some sense of.  We seem to have come a very long way from "performance menu not showing up".

Kevin

RK11111111111, 2024-01-26:
I think it's a "feature" of the Alesis.  The buttons 3 and 4 seem to change the channel of the keybed irregardless to how they are mapped to the minidexed.  For example, I mapped buttons 1 and two to select and back, and the keybed continued to change channels.

If I map buttons 3 and 4 to prev/next, it kind of makes sense at the top level as it toggles through the TG.  Though, if I try to edit a voice it doesn't work, as it keeps jumping channels.

I'm tempted to put TG one in omni mode, then map the pads to higher channels.

I was thinking I could map the buttons 1/2 to performance pgmup/down, then leave buttons 3/4 unmapped, but I can't seem to get the performance up/down to work.  What exactly does pgup and pgdown mean.

diyelectromusic, 2024-01-26:
> I was thinking I could map the buttons 1/2 to performance pgmup/down, then leave buttons 3/4 unmapped, but I can't seem to get the performance up/down to work. What exactly does pgup and pgdown mean.

They should increase or decrease the "program" number.  So if PCCH is set to a MIDI channel, then Program Change messages and these buttons should work on performances.  If set to disabled, then Program Change and these buttons should act on voices for the selected TG (which itself can be changed with the TGUp/Down buttons if configured).

Kevin

RK11111111111, 2024-01-28:
Ah, I can have minidexed send the correct midi channel back to the Alesis keyboard: https://github.com/tmick0/alesisvsysex/issues/5

closed 2024-01-31 by RK11111111111