probonopd/MiniDexed #599: Selecting banks + main volume + MIDIAutoVoiceDumpOnPC

soholt, 2024-01-10:
I am trying to make a midi controller https://github.com/soholt/MiniDexed-Lemons

If I send 0xb1 20 00 to change the bank to 000000_rom3a.syx, bank 000001_rom3b.syx is loaded displayed on LCD (from getsysex.sh)

Main Volume also do not seem to function

Also MIDIAutoVoiceDumpOnPC=1 seems to work, can I trigger it manually(needed to update ui)?

How do I upload performaces? I can send individual voices, but how do I upload the rest of the bulk params?

Thanks

diyelectromusic, 2024-01-11:
That is because the files are numbered according to MIDI expectations of what the user should see (i.e. 1 to 128) but the MIDI messages are numbered according to how these should be translated "on the wire" so 0 to 127.  Hence MIDI bank msg 0 is file 00001, MIDI msg bank 1 is file 000002, etc.

btw the full sequence is meant to send both LSB and MSB I believe, so both CC 0x00 and CC 0x20.

Note sure why MIDIAutoVoiceDumpOnPC isn't working, pretty sure it worked last time I checked, so I'll have a quick look.  Both channel volume (Bx 07 xx) and "master volume" SysEx messages ought to work too...

We can't upload performances atm i'm afraid.  We don't have any SysEx link to those at present.

Kevin

soholt, 2024-01-11:
Hi Kevin,

(In my opinion, this is madness. If I ask for bank 0, I expect bank 0 and then the display can show the bank index+1 to the user, or any other number I choose. So bank 000000_rom3a.syx seems to be a waste of space as it will never be loaded unless I send -1, to load 0..) Will update getsysex.sh to count from 1

On bank change, I send 3 msgs, bank CC MSB and LSB + PC to load the voice number to match the currently selected voice id, will try sending it as one packed msg (midi debug msgs can be seen in a browser console F12)
[image]

Individual Channel Volumes work, but not the main Volume that is controlled by sysex.

Just learned that Chromium does not like local ssl web sockets, please use Firefox.

Thanks, Gin

diyelectromusic, 2024-01-11:
> (In my opinion, this is madness. If I ask for bank 0, I expect bank 0 and then the display can show the bank index+1 to the user, or any other number I choose.

That's just how MIDI works.  See https://github.com/probonopd/MiniDexed/discussions/456#discussioncomment-5543531

It is quite clear on the subject: when "telling" the user, use 1-indexed ranges.  On the wire, use 0-indexed ranges.  Of course, really what you should be saying in your statement here is "I am a person, so I'm asking for Bank 1 and I don't care what magic goes on underneath to select bank 1".

Since the changes April last year (see https://github.com/probonopd/MiniDexed/discussions/482) bank filenames have started from 00001 to reflect the MIDI specification so that filenames match what any tool will be showing the user so the numbers and names will match.

Kevin

diyelectromusic, 2024-01-11:
Right - yes, that is a bug in getsysex.sh :)

soholt, 2024-01-16:
> > (In my opinion, this is madness. If I ask for bank 0, I expect bank 0 and then the display can show the bank index+1 to the user, or any other number I choose.
> 
> That's just how MIDI works. See [#456 (reply in thread)](https://github.com/probonopd/MiniDexed/discussions/456#discussioncomment-5543531)
> 
> It is quite clear on the subject: when "telling" the user, use 1-indexed ranges. On the wire, use 0-indexed ranges. Of course, really what you should be saying in your statement here is "I am a person, so I'm asking for Bank 1 and I don't care what magic goes on underneath to select bank 1".
> 
> Since the changes April last year (see #482) bank filenames have started from 00001 to reflect the MIDI specification so that filenames match what any tool will be showing the user so the numbers and names will match.
> 
> Kevin

Thank you, I will read it soon