probonopd/MiniDexed #613: There are weird actions happening in my fork of MiniDexed!

diyelectromusic, 2024-02-05:
I'm not sure what has changed, but it wasn't me (honest!) - but whenever I sync up my fork of MiniDexed I'm now getting actions running and errors appearing.  See: https://github.com/diyelectromusic/MiniDexed/actions

Has something changed in the main build.yml/etc that would cause this? I'm wondering if it is relates to this (it seems that it might line up timing wise): https://github.com/probonopd/MiniDexed/commit/d99b9862bb47ebdbf17d3448adb599bcb343e5ce

If so - can we discuss these kinds of things before they just get applied, so we know what to expect?

Thanks,
Kevin

probonopd, 2024-02-05:
Hi @diyelectromusic 

The purpose of https://github.com/probonopd/MiniDexed/blob/main/.github/workflows/pr-comment.yml is to add a comment to merge requests with the direct download link of the continuous build. The details are described at https://nightly.link/.

For commits other than pull requests, this is simply skipped, e.g., here:
https://github.com/diyelectromusic/MiniDexed/actions/runs/7788117982/job/21236852691

I haven't seen it work properly yet, either. But I do think it would add value if we can get it to work properly.

diyelectromusic, 2024-02-05:
The issue isn't what it is meant to do in your main repository, the issue is that every update in my own fork now attempts an automatic build (see the link) and fails.  Every time I sync I now get an error email.  This has only started happening recently and I don't know why - as far as I know, I've not changed anything in my own fork...

probonopd, 2024-02-05:
> I haven't seen it work properly yet, either.

Spoke too early. Here it is in action:

https://github.com/probonopd/MiniDexed/pull/614#issuecomment-1927958320

That is what it is intended to do.

---

But looking at your log more closely, what actually seems to fail is the uploading of a release to GitHub Releases in your repo.

https://github.com/probonopd/MiniDexed/blob/891a57809c098d77374558baa49141144771eb14/.github/workflows/build.yml#L83-L88

I can see two ways around this:
1. You allow the action to succeed in your repo (so that it can upload to GitHub Releases in your repo whenever you do a build in your main branch), by going to Settings -> Actions -> General -> Workflow permissions and setting it to "Read and write permissions", or
2. We find an `if` statement that makes this step run only in the https://github.com/probonopd/MiniDexed/ repository, but not in forks of it

Which one do you prefer?

diyelectromusic, 2024-02-05:
My question is - why is the action running at all?  I'm pretty sure the fork never ran actions before!? :)

Kevin

probonopd, 2024-02-05:
Now, that is a great question, to which I don't know the answer. Maybe GitHub has changed some default? Not sure.

Settings -> Actions -> General -> Actions permissions -> "Disable actions" to switch them off.

diyelectromusic, 2024-02-05:
Ok, happy to blame GitHub (or maybe I did turn something on without knowing I did!?  You know me and GitHub - we get on so well...) and close this then :)

Thanks!
Kevin

closed 2024-02-05 by diyelectromusic