probonopd/MiniDexed #624: Sostenuto pedal does nothing

Kirtai, 2024-03-11:
I just tried the latest release with my dpiano and while the sustain pedal works the sostenuto one doesn't.

Any thoughts on adding support for this?

The MIDI CC# for sostenuto is 66.

diyelectromusic, 2024-03-14:
Ok, so I've had a bit of a look at this and I'm not sure this is implemented in the Synth_Dexed synth engine MiniDexed uses.

I've found references online to sostenuto being "key hold" which can be set to be controlled by the second foot controller (FC2) in a DX7ii:

[image]

But can't see where this maps into Dexed itself - I'm not sure it is part of the original DX7?  There are a number of API calls in Synth_Dexed to set what the foot controller will do for voices (choosing combinations of pitch, amplitude, EG and the range) and there is a global sustain function.  But I can't see a mention of sostenuto or key hold in the original DX7 manual and I'm not sure the original DX7 recognised CC66 either from the MIDI implementation chart I've found - again I'm wondering if this might be a DX7ii thing...

Kevin

Kirtai, 2024-03-14:
"Key Hold" sounds like the right thing though it's sad that it's not in the original DX7. :(

Kirtai, 2024-04-01:
Should I suggest this to the upstream Synth_Dexed or the original Dexed?

diyelectromusic, 2024-04-01:
Hmm, good question.  Not sure.  I guess ultimately it would have to be in Synth_Dexed regardless even if the original change is further upstream, so I'd probably start there.

I can see that each note has a "sustain" flag but that seems to always be set according to the voice's global sustain status.  I don't know how the two would interact so it might be necessary to introduce a "hold" flag for notes too to allow either sustain or hold to hang on...

I don't follow the Dexed code as much as Synth_Dexed, but I couldn't obviously see it implemented there either.  It certainly didn't look like it processed CC 66, only CC 64 - this is in PluginProcessor/processMidiMessage, which appears to have most of the logic that became dexed.cpp in Synth_Dexed.

Given how the code is quite diverged anyway, it could perhaps be added to Synth_Dexed and not worry about the original...
I don't know how much time Holger has to think about such things in Synth_Dexed these days though...

Kevin

soyersoyer, 2024-11-07:
https://codeberg.org/dcoredump/Synth_Dexed/pulls/7

probonopd, 2024-11-07:
Thanks @soyersoyer. As soon as it is merged in https://codeberg.org/dcoredump/Synth_Dexed, we shall update MiniDexed to use the updated Synth_Dexed version.

dcoredump, 2024-11-08:
It's merged in Synth_Dexed.

Thanks!

probonopd, 2024-11-09:
@Kirtai please test whether the build in https://github.com/probonopd/MiniDexed/pull/750 fixes this for you. Thanks!

closed 2025-07-08 by probonopd