probonopd/MiniDexed #626: [Feature Request] Option to have TG UP/Down stay on same submenu

coax75ohm, 2024-03-18:
Currently, when using TGUp/Down it takes you up to the first level submenu for that TG and shows "Voice". I think it would be more convenient if it would stay in the same submenu position. For example, if you're in the volume setting, hitting TGUp would step through each of the TG volume settings so you could quickly set your mix levels across all the voices without having to drill down the menu for every one. Same for Pan, Reverb-Send, etc.

If you are in Effects or Performance, it could possibly just ignore the button press, or keep the current behavior.

probonopd, 2024-03-25:
Are you aware of https://github.com/probonopd/MiniDexed/wiki/Menu#shortcuts?

coax75ohm, 2024-03-26:
Oh, that sounds like what I want, but the build I'm working on uses 6 buttons without an encoder. Can it be configured to hold one of the buttons (select) then use prev/next? I played around some, but couldn't get it to work.

probonopd, 2024-04-27:
I am using a configuration with 5 buttons without an encoder: Up, Down, Left, Right, OK. Arranged like this:

[image]

Works well with the current code.

closed 2024-04-27 by probonopd

coax75ohm, 2024-05-08:
Could you share your config? I haven't been able to get shortcuts to work with a button only build.

BTW, here's mine
[image]

From the top, the buttons are: Program Up/Down, Prev/Next, Select/Back.

    #GPIO Buttons
    ButtonPinPgmDown=4
    ButtonActionPgmDown=click
    ButtonPinPgmUp=17
    ButtonActionPgmUp=click
    ButtonPinPrev=27
    ButtonActionPrev=click
    ButtonPinNext=22
    ButtonActionNext=click
    ButtonPinSelect=23
    ButtonActionSelect=click
    ButtonPinBack=24
    ButtonActionBack=click`

probonopd, 2024-05-09:
Posted my config above. Whether the shortcut works with buttons only... not so sure actually. Would probably really need the "OK" button for that. And I'm not sure whether our current code reads simultaneous button presses correctly. To be tested.