probonopd/MiniDexed #628: MIDIButtonNotes ignores MIDIButtonCh

coax75ohm, 2024-03-21:
I tried using the MIDIButtonNotes feature, but it seems it listen to all MIDI channels, not what is set by MIDIButtonCh. This makes it useless unless you're trying to make glitch music.

diyelectromusic, 2024-03-22:
Hmm.  I was pretty sure that worked last time I tried it!  I wonder if I've broken it recently...

From the code I think the check is still happening... https://github.com/probonopd/MiniDexed/blob/main/src/userinterface.cpp#L373

 I'll try to give it a go over the weekend and see what's going on.  Can you paste in the appropriate section of your minidexed.ini so I try the same config as you?

Kevin

coax75ohm, 2024-03-23:
I did a bit of testing and the omni mode happens on MIDI channel 16 (should be greater than 16), channels 1 and 15 worked fine.

```
MIDIButtonCh=16
MIDIButtonNotes=1
MIDIButtonPrev=24
MIDIButtonNext=26
MIDIButtonSelect=28
MIDIButtonBack=29
MIDIButtonTGDown=31
MIDIButtonTGUp=33
```
Using a Zero 2W at the moment and don't have a mini HDMI cable yet to check the debug.

diyelectromusic, 2024-03-23:
Ah sounds like a bug in the channel handling then - thanks - I'll take a look :)

diyelectromusic, 2024-03-23:
Yep, pretty sure the culprit is this line which should be a <= not a < as it is translating from user-configured MIDI channels (1 to 16) to internal device MIDI channels (0 to 15)...

Will see if I can test it properly in a bit

diyelectromusic, 2024-03-23:
If you get a moment, can you give this a go?  Should have sorted it!

[MiniDexed_2024-03-23-6b6065e](https://github.com/probonopd/MiniDexed/actions/runs/8403073226/artifacts/1352592392)

Yes, with a HDMI display I can see channel 16 was coming up as OMNI.  Also, the log message was reporting the "internal" number not the user-set number (1-16) so I've updated that too.

Thanks for reporting :)

Kevin

probonopd, 2024-03-25:
Merged the fix mentioned above. Please let me know in case this issue persists, then we can always reopen it. Thanks!

closed 2024-03-25 by probonopd

coax75ohm, 2024-03-26:
Thanks, though I don't really need it ;) Was just a temp thing while I was working on building the hardware.