probonopd/MiniDexed #644: miniDexed on Raspberry Pi2 and the Compressor Mystery

Banana71, 2024-04-20:
The Raspberry PI2 has problems with some performances.
When the compressor and all TGs are used, crackles occur. The error is present across versions and not just on the current one. I haven't had an RPI 2 in operation for a while and I only noticed it now. 🐼 
The easiest solution is to turn off the compressor. The compressor is only used as a limiter anyway and is therefore not relevant to the sound.
Is it possible to permanently deactivate the compressor for the RPI2 from a programming perspective?

Banana71, 2024-04-21:
I checked all performances to see if they worked smoothly with an RPi 2 and adjusted them accordingly.

diyelectromusic, 2024-04-21:
I'm sure it will be - I'll take a look at some point.

By the way, do you know about the profiling debug output?  I believe it can be used to see how much is being spent generating samples, so we can see how close a build/hardware combination is to maxing out.  I've not really used it myself, and it does incur other overheads with the continual printing, but that might be useful when investigating things like this - or at least to help us quantify things.

Kevin

Banana71, 2024-04-21:
My Raspberry Pi 2 is in my Roland MKB-300 and I can't access the HDMI interface.
Maybe it would be better to point out the problem with the compressor and the RPi 2 in the WIKI. Problems only arise in exceptional cases and can then easily be avoided. For some sounds I made the voices extra loud to get the compressor to work, so I want to check the effects again when the compressor is off.