probonopd/MiniDexed #646: Option to have reverb (or other future effects) in mono mode

baconmonsta, 2024-04-24:
I use Minidexed as 2 synth engines TG1 & TG2" panned hard left and right. The output channels (left & right) go to different destinations in my music setup. If I use reverb on either of the engines, the reverberated sound will start leaking out from the other channel as well, rendering it useless for my use case.

Therefore I'd propose having the option to set reverb into mono-mode (i.e. no stereo widening) to keep the stereo image clean.

probonopd, 2024-04-24:
> The output channels (left & right) go to different destinations in my music setup

In advanced setups like this, don't most people use external effects processors/"guitar pedals" for the effects anyway?

baconmonsta, 2024-04-24:
The upside of having this, would be to have the option of a midi-controlled reverb in the composition. But you're right, pedals will do the trick as well.

That being said, I really love this project. I have been looking for a good and compact DX7-like fm synth for a while, and never in my wildest dreams imagined that the best solution would be a diy project. Thank you!

commit afa72d21 (2024-05-30): Attempt to fix #646 (#649)

* Attempt to fix #646

https://github.com/probonopd/MiniDexed/issues/648

* SetVoiceName() suggested by @diyelectromusic
closed 2024-05-30 by probonopd