probonopd/MiniDexed #647: Enhanced midi visual feedback and display modes

baconmonsta, 2024-04-24:
On Mutable Instruments Shruthi [https://www.soundonsound.com/reviews/mutable-instruments-shruthi-1](url) there is a flashing note symbol displayed on the screen whenever the device is receiving note information via midi, and a flashing tilde symbol when it's receiving midi cc.

Would this be possible to be implemented on minidexed as well?

Another cool addition would be to have different display modes, which could be cycled through. There would be display states, where several TG:s are displayed at the same time, and whether they are receiving midi or cc info.

probonopd, 2024-04-24:
> a flashing note symbol displayed on the screen whenever the device is receiving note information via midi, and a flashing tilde symbol when it's receiving midi cc

Agree that this would be useful.

> Another cool addition would be to have different display modes, which could be cycled through.

While useful for power users, I imagine this might significantly complicate both the code and usage.

baconmonsta, 2024-04-24:
Sure, don't see why a led wouldn't do the trick as well.

The display modes are not so important. 

Generally speaking though, it would make things easier for not-so-skilled builders (like me) to have a bit more information available for troubleshooting. Even just a message in the console (via hdmi) for first received midi note and first time a TG is playing a sound

probonopd, 2024-04-25:
You know that you can enable HDMI logging in the minidexed.ini file? Then it will show received MIDI messages. It should be disabled when not needed though, because it can worsen the sound.

baconmonsta, 2024-04-26:
I did not know that... Well, that definitely makes debugging easier!

crunchysteve, 2025-01-12:
Dunno if this is within scope (or even possible) but it seems a missed opportunity to not be using the HDMI to emulate a "front panel" on one of the many RPi compatible touch screens. Quite aware that it's probably the first question noob to the project asks, but haven't seen it mentioned. I've finally found a use for my old RPi 3 B+ other than driving my oscilloscope. (Which my Mac does better, anyway.)

probonopd, 2025-01-12:
What kind of front panel would you like to see emulated? The DX7 one is (let's say) "not universally loved" and if you are looking for that kind of thing, https://github.com/chiaccona/VDX7 might be a better fit.

Then, there is this kind of monstrosity

[image]

Imagine this 8 times because MiniDexed has 8 instrument instances.

Could that be done in software on a RPi with a touchscreen? Certainly. But then, would it be much different from running https://asb2m10.github.io/dexed/ to control MiniDexed, which is already possible today?

crunchysteve, 2025-01-12:
Um, something somewhere in between? OK, I asked a noob question, sorry. I was imaging something like the TX816 front panel, but maybe as 2 rows of 4 on a common 720 or 1080 hdmi screen with a usb touch layer. If it's beyond project scope, that's cool. Anyhoo, thanks for answering. Will still be looking into getting a rig running as is for my aging YouRock guitar controller. Hence, DAW plugins, no. I have so many of those in Logic Pro X, I need to go DAWless.

probonopd, 2025-01-12:
I think the best way to go about this is to decouple the tone generator (MiniDexed) from the front panel ("programmer") and have the two communicate over MIDI. This way, everyone has complete freedom to make front panels while MiniDexed can stay largely unchanged.

For example, check out this bad boy:

[image]