probonopd/MiniDexed #654: Send "Bank Select" and "Program Change" over serial MIDI

AndGoGitHub, 2024-05-15:
Hello, 
It might be handy to have the ability to receive "Bank Select" and "Program Change" codes when selecting them through the MiniDexed user interface so you can record them into a DAW or keep in sync with a custom hardware controller.
Many thanks!

probonopd, 2024-05-25:
Need to think through what the correct behavior is when voices are changed vs. when performances are changed. Possibly need to take he [configuration items](https://github.com/probonopd/MiniDexed/wiki/Files) `MIDIAutoVoiceDumpOnPC`, `ExpandPCAcrossBanks` and/or `PerformanceSelectChannel` into consideration here.

probonopd, 2024-05-25:
On 1.:

`MIDIAutoVoiceDumpOnPC` allows users to synchronize MiniDexed with external MIDI devices by sending SysEx dumps whenever a voice is changed, ensuring that the connected device accurately reflects the changes made within MiniDexed. One use case might be to capture voices into empty slots on an attached device (e.g., other synthesizer or host computer). 

However, if Bank Select and Program Change messages are also sent alongside SysEx dumps, there's a risk that the dump could overwrite voices in unintended slots on the external device. This is because Bank Select and Program Change messages determine the target location. If these messages were triggered concurrently with SysEx dumps, there's a possibility of conflicting instructions, potentially leading to patches being stored in incorrect slots or overwritten unintentionally.

probonopd, 2024-05-25:
On 2.:

The DX7 synthesizer has a total of 32 available slots in its internal memory for storing voices. Each slot can hold one voice patch, which consists of parameters defining the sound characteristics such as waveform, envelope settings, and modulation parameters. These slots are often organized into banks, with each bank containing a set of 32 slots.

When sending Bank Select and Program Change messages, MiniDexed would need to reverse or adjust the mapping to ensure that the sent out information aligns with the expected behavior of a DX7. This adjustment is crucial to maintain compatibility with other MIDI devices and software.

diyelectromusic, 2024-05-25:
I must admit I wasn't quite understanding the use-case.  Should it be recording Bank/Voice changes in real time in the DAW?  I can't quite see the MiniDexed UI really being much good for this :)

Is the real question how to get the numbers out of MiniDexed for inputting into a DAW manually?  At present this is down to knowing what files you have on your system.  In performance mode, performance bank/patch numbers are displayed, but the presentation is quite inconsistent between performances and voices.

Agree with the comments above too - there could be some complex interactions if we don't think it through...

But presumably the original DX7 did send out patch changes over MIDI so it could control a TX7 or TX816, so it isn't unreasonable to think about it in MiniDexed too, but as I say, the UI isn't really supportive of doing it in a useful way at the moment in my view.

Kevin

AndGoGitHub, 2024-05-27:
Hello to all, many thanks for the answers!
Just to poin out some ideas:
I think that the "MIDIAutoVoiceDumpOnPC" option (it seems I haven't seen it
on the wiki) is far too much to only keep track of the location of PROGRAM
CHANGE plus any BANK SELECT recalled from the minidexed UI (NOT PERFORMANCE
SELECT!).
I don't think there could be synchronism problems because you could always
put a configuration option for sending these two parameters (program change
and bank select) and excludes (automatically or not) the other one.
In any case to clarify, I'm just making an hardware controller to send
direct program changes and banks select by push buttons, like in old style
keyboard synthesizer and the only problem is that if someone (the user)
change those parameters from the minidexed UI (LCD + Encoder) it will be
not reflected back to the push button's leds.
Another question is if the minidexed midi implementation table is missing
some added new features, particularly in the send section that is (to me)
currently missing.
Many thanks for your time and your precious work!

Andrea

diyelectromusic, 2024-05-28:
> In any case to clarify, I'm just making an hardware controller to send direct program changes and banks select by push buttons, like in old style keyboard synthesizer and the only problem is that if [...]

Yes, this was the issue I had with my pseudo TX816 build too.  I don't really have a useful solution in mind to it at the moment, but I don't think sending MIDI program/bank change is the answer - that does nothing for other parameters, such as volume, and so on and it is easy to get into a mess with MIDI channels and performances and so on.

I still wonder that the best option is to implement effectively UI SysEx remote control, like I used in my test code for a dual MiniDexed.  That would allow something else to become the UI on behalf of a MiniDexed.  But that is a little complicated too.

Kevin

probonopd, 2024-05-30:
Can't MiniDexed already be entirely remote controlled using [MIDI Button Navigation](MIDI Button Navigation)?

diyelectromusic, 2024-05-30:
Mostly, but it still needs a screen of course.

The SysEx remote control I had in my dual setting allowed a more integrated UI experience and removed the problem of keeping things in sync if (as in this case) there is a microcontroller handling a UI instructing the MiniDexed what to do in parallel with the MiniDexed itself. Its kind of like the SysEx gives a remote "command line" type action compared to the MIDI buttons giving a remote "app/desktop/web" type action.  Well, not really, but that is the closest comparison that springs to mind at the moment :)

... anyway, that is quite an advanced use case really.  I'd still like to have it as an option one day though because MiniDexed is definitely something people build into other things and this would be a really neat thing to be able to do!

Kevin