probonopd/MiniDexed #698: Help Needed: ST7789 Display Not Working with Pi Zero 2 and PCM5102A DAC Wiring Suggestions 😅.

EphraimElgrabli, 2024-08-10:
Hi there! First off, I want to say that I love this project—great work!

I’m trying to get everything set up using the following components:

I2S PCM5102A DAC: [AliExpress Link](https://www.aliexpress.com/item/1005006816695590.html?spm=a2g0o.order_list.order_list_main.11.2440586aKplxuO&gatewayAdapt=glo2isr#nav-specification)
ST7789 Display: [AliExpress Link](https://www.aliexpress.com/item/32858069771.html?spm=a2g0o.order_list.order_list_main.17.2440586aKplxuO&gatewayAdapt=glo2isr#nav-specification)
Raspberry Pi Zero 2
Here’s the pin mapping I’ve used for the display:

ST7789 Display:

GND: Pin 6 (Ground)
VDC: Pin 1 (3.3V Power Supply)
SCL: Pin 23 (GPIO 11, SPI0 SCLK)
SDA: Pin 19 (GPIO 10, SPI0 MOSI)
RES: Pin 22 (GPIO 25)
DC: Pin 24 (GPIO 8, SPI0 CE0)
BLK: Pin 35 (GPIO 19)
I’ve configured the ini file for the display like this:

ini:
'''
SPIBus=0
ST7789Enabled=1
ST7789Data=19
ST7789Select=1
ST7789Reset=22
ST7789Backlight=35
ST7789Width=240
ST7789Height=240
ST7789Rotation=0
ST7789SmallFont=1

LCDColumns=15
LCDRows=2
'''

However, when I power on the Pi Zero 2, nothing appears on the screen. Is this normal, or could there be an issue with my pin mapping? This is my first time working on an embedded project and playing around with wiring, so I’m not entirely sure where I might have gone wrong (though I’m having a lot of fun with it!).

Additionally, could you suggest the correct wiring map for the PCM5102A DAC?

Thank you so much in advance for your help—I really appreciate it!

craigyjp, 2024-08-10:
It doesn't have a chip select pin. There are two types of ST7789, with and without. Maybe it needs a cs pin, I cannot say for sure. I use ST7789 displays with a cs pin in my other projects.

I believe you can mod these displays to replace the back light pin for cs.

EphraimElgrabli, 2024-08-10:
Hi @craigyjp,

Thank you so much for the quick reply! I really appreciate it.

Could you guide me on how to make this modification? Unfortunately, I’m more of a software guy and don’t have much experience with hardware. Any detailed instructions or resources you could share would be incredibly helpful.

Thanks again for your support!

craigyjp, 2024-08-10:
Really if you don't have hardware experience it's a tricky job, I did it once following a website guide as I originally bought the same ones without a cs pin. You need to cut traces and link traces to different places after lifting that silver foil ribbon, not for the faint hearted.

Just wait to see if it can be used without a cs pin, maybe you need a config change. I know on teensy you can use them without a cs pin by setting the config differently, maybe it's the same here.

EphraimElgrabli, 2024-08-10:
I think I’ll just buy another screen instead. Do you have any recommendations for a different one?

By the way, I’ve seen your work on YouTube, and it’s absolutely amazing—truly great job!

craigyjp, 2024-08-10:
One other question, is the ST7789 supported? I know the SSD1306 is, but I had not heard about any other graphics displays, it's a while since I've kept up with the project.

EphraimElgrabli, 2024-08-10:
Yes, it is, at least according to the wiki. Here’s the link: [MiniDexed Wiki - SPI ST7789 Displays](https://github.com/probonopd/MiniDexed/wiki/Hardware#spi-st7789-displays).

craigyjp, 2024-08-10:
Yeah, just read it, fairly new, I will stick with my 16*2 OLED

EphraimElgrabli, 2024-08-10:
So, do you recommend getting a different screen instead? If possible, could you send me a link to one that will definitely work?

Also, is the DAC I bought suitable for this project?

craigyjp, 2024-08-10:
It's up to you, I used a 1602 character OLED, works perfectly well, but that was before any graphical displays were even supported. I guess now I would try to use an ST7789 because they are a lot cheaper. 

Like I said it's been a while since I've touched this project especially with the war raging in my country most of my gear is packed away.

So I know my OLED was a winstar 1602, from what I remember it uses 4 bit mode so not a lot of connections required. I thought it was i2c adapted, but again that came later.

Can't remember what audio board I used, just a cheap generic 1502 from AliExpress

EphraimElgrabli, 2024-08-10:
ok, thank u so much!

Banana71, 2024-08-11:
Hello Ephraim, here I compared the different sizes of SSD1306 compatible OLEDs.
https://github.com/probonopd/MiniDexed/discussions/656#discussion-6732328

EphraimElgrabli, 2024-08-11:
thank u so much!

closed 2024-08-11 by EphraimElgrabli