probonopd/MiniDexed #712: USB MIDI Controller Connection Error on Raspberry Pi Zero W 2

EphraimElgrabli, 2024-09-01:
When attempting to connect a USB MIDI controller to my Raspberry Pi Zero W 2, I encounter the following error in the HDMI output
"dwroot: previous attempt to initialize device failed"

### Environment
Device: Raspberry Pi Zero W 2
MIDI Controller: M-audio Keystation 49 (listed as a working device in the list)

minidexed midi config:
MIDIButtonCh=1
MIDIButtonNotes=1
MIDIButtonPrev=98
MIDIButtonNext=99
MIDIButtonBack=93
MIDIButtonSelect=100
MIDIButtonHome=0
MIDIButtonPgmUp=96
MIDIButtonPgmDown=97
MIDIButtonTGUp=95
MIDIButtonTGDown=94

thanks for the help in advance.

diyelectromusic, 2024-09-01:
I think we'd probably need to see all the log messages on the display to see what is going on.  In particular anything related to that "previous attempt" :)

btw - have you seen the comments on the wiki about various cmdline.txt settings that might affect USB?

Kevin

EphraimElgrabli, 2024-09-01:
yes I have seen and tried both usb speed full and not full and didn't work either way
I sadly disconnected the pi from power and pack it since it's night here already, I'll send you a picture tomorrow. the error appear when connecting the midi to the usb port.

EphraimElgrabli, 2024-09-02:
I've resolved the issue. The problem was that I had incorrectly soldered the data+ and data- wires. After resoldering them in the correct orientation, everything is working properly. Thank you for your assistance!

closed 2024-09-02 by EphraimElgrabli