probonopd/MiniDexed #713: Fast GPIO button clicks go undetected

miotislucifugis, 2024-09-01:
Ive recently completed a minidexed build that uses mechanical keyboard switches for GPIO buttons and Im finding that with these switches that allow faster click action (vs the standard "clicky" tiny pcb switch) ,  many times the quick taps go undetected and it requires sort of a longer deliberate press to be recognized.    I assume that this is due to a switch debounce delay in software.   Is there any way to customize this? ... or can it be added as a custom setting, like the double click and longpress timeouts?

probonopd, 2024-09-01:
Possibly we can lower `BUTTONS_UPDATE_NUM_TICKS` and/or `DEBOUNCE_TIME` in https://github.com/probonopd/MiniDexed/blob/main/src/uibuttons.h.

probonopd, 2024-09-01:
Please test the build from https://github.com/probonopd/MiniDexed/pull/714 - does it improve/solve the issue?

diyelectromusic, 2024-09-01:
If the issue is detecting short presses in the first place, rather than repeated presses, then it is probably BUTTONS_UPDATE_NUM_TICKS that needs decreasing.  This would however mean that more time is spent checking buttons rather than processing audio, so we'd need to check performance on lower-end devices too to ensure audio is still ok.

DEBOUNCE_TIME I'd expect would affect how quickly a second press of the button is recognised, e.g. in a double click.

I don't recall how we came up with those numbers - I expect it was an "initial finger in the air" guess and then see how it works :)

Kevin

miotislucifugis, 2024-09-01:
(copied from my response in https://github.com/probonopd/MiniDexed/pull/714) Wow, Thanks!   This seems to be working great.  Even the fastest, lightest taps get registered.     However, Im finding that if i do a couple of fast successive taps, like 3 quick ones in a row to go back 3 steps,  only only one press gets identified....but I guess that this is a limit of the double tap timeout.

probonopd, 2024-09-03:
@diyelectromusic agree to merge this?

miotislucifugis, 2024-09-15:
this can be closed.  Everything is working fine now

closed 2024-09-15 by miotislucifugis