probonopd/MiniDexed #716: Confused with wiring on PiFi HIFI DAC+ 2.0

Potajito, 2024-09-04:
Hi!
I saw that this board was working, but I'm too dumb to make it work...
Basically, I think my problem is that I don't know where SCK is, because the rest of the wiring (attached) seems ok? 
Right now I have a Raspberry zero 2w, a LCD1602 display and a rotary encoder, all working, also receiving midi signals, but so far no sound (red led on the dac, for what is worth).
Any ideas?
Thanks!
[image]

diyelectromusic, 2024-09-04:
I2S usually only requires CLK/BCLK (GPIO 18), LCK/FS (GPIO 19) and DOUT (GPIO 21). I'm not sure where SCK has come from unless it is another name for the main clock (CLK/BCLK).

Regardless, it should all be on the right GPIO pins on that header anyway, so shouldn't it just plug right in?

What I'm not quite sure about is if this board needs some I2C configuration too - if it is based on the HiFiBerry boards, then it might... https://www.hifiberry.com/docs/hardware/gpio-usage-of-hifiberry-boards/

Kevin

Potajito, 2024-09-04:
> I2S usually only requires CLK/BCLK (GPIO 18), LCK/FS (GPIO 19) and DOUT (GPIO 21). I'm not sure where SCK has come from unless it is another name for the main clock (CLK/BCLK).
> 
> Regardless, it should all be on the right GPIO pins on that header anyway, so shouldn't it just plug right in?
> 
> What I'm not quite sure about is if this board needs some I2C configuration too - if it is based on the HiFiBerry boards, then it might... https://www.hifiberry.com/docs/hardware/gpio-usage-of-hifiberry-boards/
> 
> Yes, looks like the PCM5122 does need a I2C configuration interface which we don't directly support iirc.... https://www.ti.com/document-viewer/PCM5122/datasheet#application-and-implementation-lit57966096/lit57966096
> 
> Update: Yes we do support that - you'll probably need to set `DACI2CAddress` in minidexed.ini and make sure the I2C bus is also connected (GPIO 2 and 3).
> 
> btw - queries for help like this aren't really issues, so you might get more people responding if you post in the discussion area.
> 
> Kevin

Ahhh! Thanks! Sorry for the issue, I didn't remember the project had a help forum (and I've posted there before, duh). 
I'm going to try all this now, thanks!
Just one last thing, do I set `DACI2CAddress` as "1" or any other value? I guessed that could be an issue, but couldn't find much about it.
Thanks!

diyelectromusic, 2024-09-04:
It needs to be set to whatever address the board is using, but I can't find any information about the board online.

If it is matching a HiFiBerry plus it /might/ be 0x4d (according to the dt overlay file) but you might have to boot into a RPi OS and run i2cdetect or hook up an Arduino and run an i2c scan...

Kevin