probonopd/MiniDexed #718: Midi - Programm change doesn't Change patchname at display

Gitarrenzupfer, 2024-09-07:
Hello, i just got started to fiddle with this awesome project. I took a raspberry 3 and everything worked quite good, but if I send a program change message in Performance-Mode, the sound is changing, but not the patchname/number on the display. 

Any ideas/ hints?

greetings 
Gitarrenzupfer

diyelectromusic, 2024-09-07:
If I understand you correctly, that is a consequence of how performances are chosen in the menu.  The "[L]" which indicates which performance is loaded, will change but it doesn't automatically step-through the display to the new chosen performance.

There is a whole pile of considerations about what the display ought to show when MIDI messages are received that we need to work through at some point.

So this isn't an issue as such, it is as currently designed.  As I say, assuming I'm understanding you correctly.

Kevin

Gitarrenzupfer, 2024-09-08:
Hello Kevin,

Thanks for your quick reply. I understand, it is not an error or a bug, but "by design". I would be very pleased if it were changed in a software update and thus had the same behavior as all the other synths that I own or know.

To be honest, I don't see any point or advantage in the current behavior of the display when receiving MIDI program changes.

Regards

diyelectromusic, 2024-09-08:
> To be honest, I don't see any point or advantage in the current behavior of the display when receiving MIDI program changes.

"There is a whole pile of considerations about what the display ought to show when MIDI messages are received that we need to work through at some point."

I didn't say I agreed with the behaviour either - it's on the todo list ;)

(refs: https://github.com/probonopd/MiniDexed/issues/647 https://github.com/probonopd/MiniDexed/issues/549 https://github.com/probonopd/MiniDexed/pull/604 https://github.com/probonopd/MiniDexed/discussions/659 and probably quite a few more - as I say, it's on the todo list...)

Kevin

PR #887 (2025-04-25): Remove [L] if PerformanceSelectToLoad=0 (the new default)
Remove `[L]` next to the currently loaded performance if  `PerformanceSelectToLoad=0` (the new default), because it is not needed then; making the behavior more consistent with voices.

Closes #718
commit d209d39e (2025-04-25): Remove [L] if PerformanceSelectToLoad=0 (the new default) (#887)

Closes #718
closed 2025-04-25 by probonopd