probonopd/MiniDexed #728: With MidiDumpEnabled=1, only outputs MIDI dump to HDMI, every 31/32 input events

nhemsley, 2024-10-01:
Strange one. I think this worked a couple of days ago on the same build. 

It appears the midi keypresses are being output to hdmi every 31/32 presses.

Any ideas? Reformat?

nhemsley, 2024-10-01:
Probably caused by this: 

https://github.com/smuehlst/circle-stdlib/issues/35

probonopd, 2024-11-15:
@smuehlst do you know what's going on? Thanks

smuehlst, 2024-11-16:
@probonopd Could you explain what MIDI dump to HDMI actually does? Is it in any way related to stdio? The problem with vhttps://github.com/smuehlst/circle-stdlib/issues/35 was that `stdout` was no longer line-buffered.
@nhemsley Why do you think that https://github.com/smuehlst/circle-stdlib/issues/35 is related to this problem?

smuehlst, 2024-11-16:
Further evidence that https://github.com/smuehlst/circle-stdlib/issues/35 is unrelated:

* MiniDexed is using circle-stdlib [v15.12](https://github.com/smuehlst/circle-stdlib/releases/tag/v15.12) if I'm interpreting the submodule information correctly.
* The problem described in https://github.com/smuehlst/circle-stdlib/issues/35 was introduced in circle-stdlib [v16.3](https://github.com/smuehlst/circle-stdlib/releases/tag/v16.3) and it was fixed in circle-stdlib [v16.4](https://github.com/smuehlst/circle-stdlib/releases/tag/v16.4).

nhemsley, 2024-11-16:
@smuehlst, It is printf debugging of the midi codes coming in.

See here: https://github.com/probonopd/MiniDexed/blob/8c18e60e5133a56e7208c4ff051cc2ec80e0aaa1/src/mididevice.cpp#L141

putting a manual flush in the code linked fixes this, but probably better to upgrade to the version that fixes the line buffering bug.

probonopd, 2024-11-16:
@nhemsley could you please test the build at https://github.com/probonopd/MiniDexed/pull/758#issuecomment-2480491725?
Thanks.

nhemsley, 2024-11-17:
I gave that a try, but the PI it goes into a reboot cycle. I'll build locally and try that in a few hours.

nhemsley, 2024-12-04:
Apologies for the delay, building locally presented challenges, as its been a while, then life got in the way.