probonopd/MiniDexed #730: Cannot get device descriptor

bobylito, 2024-10-04:
Hello :) 

I have a minidexed running on raspberry pi 3B and it's working fine with my arturia MiniLab MKII. However when I try with an Oxi One, it does not seem to be recognized. I have plugged the raspberry to an HDMI output and get this error message : "Cannot get device descriptor". I could find a reference to this error message in circle https://github.com/rsta2/circle/blob/master/lib/usb/usbdevice.cpp#L222-L233 

Do you think it can be fixed on minidexed side? Oxi one? Or even on circle side?

probonopd, 2024-10-05:
Is the Oxi One USB MIDI class-compliant (i.e., does not require a driver to be installed when connected to a computer)? If not, then that is the reason. If it is, then the reason could possibly be (just a wild guess) that the USB device descriptor is larger than what MiniDexed assumes (due to the device exposing so many features)?

diyelectromusic, 2024-10-06:
If you can give me a photo of the HDMI screen output for the device, I'll have a look. I will probably have a list of statements after plugging it in that relate to the USB device enumeration.

Kevin

bobylito, 2024-10-07:
> Is the Oxi One USB MIDI class-compliant (i.e., does not require a driver to be installed when connected to a computer)?

From what I understand yes. I can plug it to any device without any software installation (mac / android). 

> If you can give me a photo of the HDMI screen output for the device, I'll have a look. I will probably have a list of statements after plugging it in that relate to the USB device enumeration.

I don't have a screenshot. I could not see any further line printed after.

diyelectromusic, 2024-10-07:
It is the few lines printed before I really could do with seeing.  When a USB device gets plugged in there are a number of diagnostic messages that might help see what is going on before the final "yes" or "no" message.

Kevin

bobylito, 2024-10-07:
Here are the screenshot before and after I plug the device:

[image]

[image]

bobylito, 2024-10-07:
Here is the screenshot after I plugged the device

[image]

diyelectromusic, 2024-10-07:
Hmm. The first device ven424-9514 seems to be registered as "SMC9514 Hub" but then I see ven424-ec00 which seems to be "SMSC9512/9514 Fast Ethernet Adapter"- so I guess this is probably the built-in hub for a PI 3... (we see something similar with a Pi 4).

I was hoping there might be more.  Weirdly it seems that circle can read the "short descriptor" but then isn't getting enough data for the full descriptor. I wonder why...

Are you able to boot into Linux and grab the output of "lsusb" and "lsusb -v"?  It is possible it has a type of audio streaming endpoint (of which MIDI is a part iirc) that isn't supported in circle.

It might be worth trying some of the usb related options in cmdline.txt (see https://github.com/rsta2/circle/blob/master/doc/cmdline.txt) but I get the feeling this might be a question for the main circle discussion area.

Kevin

diyelectromusic, 2024-10-07:
Bit of a long shot, but I see this is a fair bit of discussion about PCs not working either with certain version of the Oxi One - https://oxiinstruments.com/forum/d/224-solved-computer-cant-find-oxi-one/7

Presumably you know its fine with another computer?  It would be nice to know that Linux on a Pi 3 can see and use it fine for example.

Kevin

bobylito, 2024-10-08:
That's good leads thanks. I will try with a few other computers I have around. I will keep you updated.

probonopd, 2024-11-15:
@bobylito were you able to resolve this? Thanks.

bobylito, 2024-11-15:
No I haven't been able to solve the issue and I was not able to test with linux on the pi device yet.