probonopd/MiniDexed #731: Plugging in a USB hub causes crash and/or uninitialized root port [RPi Zero W]

SynthGal, 2024-10-08:
Howdy folks.

As the title says, plugging in a USB hub causes the device to sometimes crash/restart, and then report over HDMI that it "Cannot get device descriptor (short)" and "Cannot initialize root port".  If I unplug the hub and plug in a keyboard, it then says "Previous attempt to initialize device failed" which makes the device useless until I power cycle it.

The Raspberry Pi Zero (or RPi Zero W in my case) only has one USB port, and my MIDI keyboard doesn't have the controls for adjusting any synth parameters. I would like to be able to plug in a KORG NanoKontrol2 for doing this, but I need a USB hub for that. I've tried several different hubs and they all do this.

Any ideas?

diyelectromusic, 2024-10-09:
I'm afraid I suspect this hasn't been explored too much.  I've just had a go and found that plugging in my USB hub to a Pi Zero (V1, non-W) causes it to reboot, but then does seem to be recognised on reboot and plugging in my CME XKey seems to work ok.

Are you able to grab anything from your own HDMI display?

Kevin
[image]

SynthGal, 2024-10-11:
I grabbed a 2.4 amp power supply and now the hub + keyboard + nanokontrol works fine. Sorry to waste your time.

There is another issue where the sound rapidly stops and starts while it's printing MIDI output, but I lack the equipment to test if it's also doing it with headphone out or a DAC, and if it's a memory issue. I have an Audio DAC Shim on the way and I might have the parts to connect to the PWM lines as well. Gimme a week and I'll test with a Zero 2W with a DAC and maybe also an audio out circuit.

diyelectromusic, 2024-10-12:
No worries!  Glad you managed to work out what was going on.

Re: MIDI dump - yes, writing out to the console has a performance hit, it is really just a diagnostic, so I'm not surprised if it causes stuttering or skipping with either MIDI or audio.  It will be particularly pronounced on the single core Pi or Zero V1, which is probably only just keeping up with 8 note polyphony anyway :)

Now everything else is working that can be turned off.

Kevin

closed 2024-11-15 by probonopd

probonopd, 2024-11-17:
Just FYI, I tried 2 cheap hubs and sure enough, one is working (including hotplug) and one is not on the RPi Zero 2 W. So there may be still some quirks in the Hub handling code.