probonopd/MiniDexed #734: Minor issue with root menu

GusIsanda, 2024-10-11:
Hi, first of all thanks for all the work!

I found a minor bug in root menu and I'm not able to find the line myself. When the menu is on TG or in edit mode, you can't go one step down if it's the first item or step up if it's the last item (looping back to the last or first item), but in root menu it does.

I know it's a minimum trouble but now I need to know how it works. Thanks again!

GusIsanda, 2024-10-11:
Why I couldn't see it?  Thanks!

Banana71, 2024-10-11:
> Why I couldn't see it? 

Because there really is a very small error in the root menu :-)
The arrow "<" is missing in the "TG1" menu on the left side
and the following arrow ">" is missing in the "Performance" menu on the right side.

probonopd, 2024-11-15:
Noticed this too.

Either we should add arrows in both directions, or we should not wrap around.
Also, I think all menus should or should not wrap around (consistency), as the OP rightfully noted.

@diyelectromusic what do you think?

diyelectromusic, 2024-11-15:
People asked for it to wrap around so they could quickly get to the performance menu.  Yes, prob should have the arrows on the root.

I don't see an issue with not wrapping around on other menus - some really do have a sense of start and end for me, but the root feels different.  But could probably add it in if people felt strongly about it.

I think the bigger issues for me are the differences between selecting voices and performances, but I have quite a long list of issue/discussion numbers related to the UI, that one day I'll sit and work through...

Kevin

closed 2025-04-23 by probonopd