probonopd/MiniDexed #735: More GPIO Options and Menu screen at startup

scottfdesign, 2024-10-11:
First of all, thanks to all that contribute to this fantastic project!
Since I'm much more into just exploring and playing the great performance patches in MiniDexed, I was wondering if some more functionality could be exposed to buttons on the GPIO Pins, specifically Performance select up and down, and performance Bank select up and down. I'd love to have hard buttons in my box to quickly scroll to specific performance patches in different banks.
Also, would it be possible to set in the configuration file what menu screen to show on startup? I'd like to just start up in the Performance-Load menu. Then with the extra buttons, you can just boot up and start playing, instead of going 3 menus deep to pick your patch.

diyelectromusic, 2024-10-12:
We can do buttons for Program Up/Down and configure the system so that "Program" means "Performance" - see https://github.com/probonopd/MiniDexed/wiki/Files#buttons

But we don't support banks right now, just "Program" and TG selection, but it will wrap over from one bank to another (at least it will for voices - I think it does that for performances too!  I might need to check :))

Jumping straight into Performance or Voice selection is an of-requested feature.

More here: https://github.com/probonopd/MiniDexed/issues/718#issuecomment-2336596509

Kevin

diyelectromusic, 2024-10-21:
There is a test build of Bank Up/Down buttons and MIDI buttons now available here: https://github.com/probonopd/MiniDexed/pull/737

Kevin

closed 2025-03-10 by probonopd