probonopd/MiniDexed #773: Manipulating Master Volume makes System stop playing sounds. E-MU Xboard 25

RichyT, 2024-12-12:
System is working well and playing sounds but moving the master volume at all and everything just stops.

Powering off/replugging the keyboard doesn't seem to work at all, only a full reboot. None of the other knobs or the pitch-bend have a similar effect, just the master volume as far as I can tell.

I removed the usbspeed=full from cmdline.txt to see if that was the issue but the problem persists.

I don't have a rotary encoder/LCD to test anything with that.

RichyT, 2024-12-13:
Sure. The picture is not great as it's on a low-res display and I had all sorts of issues with glare. Let me know if you need more.

[image]

diyelectromusic, 2024-12-13:
Hmm.  I've not really done much with master volume... I am intrigued by some of the code though (and have mentioned it before: https://github.com/probonopd/MiniDexed/issues/502).

Any idea what that stream of what looks like B0 5D xx messages before it relates to?  That looks like continually sending MIDI CC message for controller 93...

I'll try to take a look and see if I can simulate something that reproduces the issue here...

Kevin

RichyT, 2024-12-13:
For this run, I just did played some random notes to ensure it was working then moved the slider a little to cause the issue. I am not sure what the other numbers are as playing the notes does not seem to correlate with values being displayed though I am not sure if there is some buffering going on. (Turning the knobs does produce an obvious correlation).

I'm not really concerned about getting the master volume working as much as having it not cause an unrecoverable error. 

I do notice that the LED sometimes displays CC along with some numbers but I'm not sure if that correlates with what you mention. I can look harder and see if it does.

diyelectromusic, 2024-12-13:
Ok.  Anyone spot the mistake here?
```
		float32_t nMasterVolume=((pMessage[5] & 0x7c) & ((pMessage[6] & 0x7c) <<7))/(1<<14);
```

It turns out that if you take the logical AND of a LSB and a MSB you get... guess what?  zero :)

So on the first SysEx master volume, it zeros everything and can never recover.

Also - no idea why that is taking 0x7C - that is grabbing bits 0b01111100 from each byte...

Kevin

diyelectromusic, 2024-12-13:
Ok, give that PR a go - hopefully that now handles Master Volume correctly!

Kevin

closed 2025-03-10 by probonopd