probonopd/MiniDexed #808: Added support for SSD1305

Sterr1, 2025-01-31:
I was commit circle display driver for support larger displays SSD1305 32 and 64 pixels height. In minidexed.ini you can set SSD1306LCDWidth=132.
I was test it. Add it to wiki.

Banana71, 2025-02-01:
Hello Sterr1,
great that you found your way to miniDexed.
Can you please provide some more information about the displays, e.g. links to the display you are using and/or photos, the Address: `SSD1306LCDI2CAddress=0x0`

Have fun making music
Peter

Sterr1, 2025-02-01:
Hello Peter. I'm newbee on Github, and not understand some things. But I have some mind for start codespace, change code and compile working project. I change included rsta2/circle driver included in MiniDexed, but when I go to rsta2/circle on github, I can't see driver circle/addon/display/ssd1306device.cpp and my changes. I don't understand why. For support SSD1305 I just change 3 strings in file code. I see in in mt32-pi project, where my display work. If I make some wrong, here my changes to circle/addon/display/ssd1306device.cpp for working SSD1305:
```

`	// Validate dimensions - only 128x32 and 128x64 supported for now
	if (!(m_nHeight == 32 || m_nHeight == 64) || !(m_nWidth == 128 || m_nWidth == 132))
		return false;
	
	const bool bIsSSD1305 = m_nWidth == 132;

	const u8 nMultiplexRatio  = m_nHeight - 1;
	const u8 nCOMPins         = (m_nHeight == 32 && !bIsSSD1305) ? 0x02 : 0x12;`
```

SSD1305 have 132 soft pixels buffer, but really have 128 visible pixels. And changed parameter COMPins. All another compatible with SSD1306. Address the same 3C or 3D selected by pin 4 in HGS128321 and protocol I2C selected by pins 17 and 18. And must be reset pin 16 connected to HIGH. It's rare display but now sales in china more display compatible with larger than 1306 0.91" size. 
0.91" too small for use. I order another for my project and test it too.

1 pic -  HGS128321 with jumpers pins 17,18.
2 pic - compare with standard 1602 display
3 pic - work with modified driver
4 pic - 3 pic - work with modified driver
On 5 picture you can see how display works on standard SSD1306 driver

My plans: in future I hope integrate mt32, SF2, JV880, MiniDexed in one SDcard switchable as in microdexed. What is your opinion - make it in your project, or make new?
And question - ftp work fine in MiniDexed and in MT32pi. Why you not update in release? 

Links: [One](https://aliexpress.ru/item/1005005575551588.html?sku_id=12000033610052919&spm=a2g2w.productlist.search_results.12.301c2553UoUj6w), [Two](https://aliexpress.ru/item/1005008432534479.html?sku_id=12000045073471562&spm=a2g2w.productlist.search_results.34.301c2553UoUj6w), [Three](https://aliexpress.ru/item/1005008410765114.html?sku_id=12000044929745595&spm=a2g2w.productlist.search_results.36.301c2553UoUj6w)
Third variant looks like my display but 64 pixels height.

[2014061208582147.pdf](https://github.com/user-attachments/files/18630362/2014061208582147.pdf)
[image]
[image]
[image]
[image]
[image]

probonopd, 2025-02-02:
Hello @Sterr1. Thank you very much. Submitted your patch for inclusion in Circle at https://github.com/rsta2/circle/pull/541. Once it is merged there and once we can use the latest Circle in MiniDexed, support for SSD1305 should appear in MiniDexed.

FTP will land in MiniDexed once https://github.com/probonopd/MiniDexed/pull/747 can be merged.

zippypinhead2025, 2025-02-23:
Hi, sadly this does not work with on zero 2w and this oled hat:
[https://www.waveshare.com/wiki/2.23inch_OLED_HAT](url)
any suggestions?
thx in advance

Sterr1, 2025-02-23:
Probonopd not compile it with suupport because rsta2 not add it in master. Try my kernel [kernel8](https://drive.google.com/file/d/1R-XAFVJ5yt-G-bHyDVuxA-1xSxhOh61e/view?usp=drive_link)

zippypinhead2025, 2025-02-23:
No dice. Sorry.

[image]

Sterr1, 2025-02-23:
`SSD1306LCDWidth=132`

?

zippypinhead2025, 2025-02-23:
there's the rub!

[image]

Thank a ton!
Спасибо!

Sterr1, 2025-02-23:
Forget switch with to 132?

zippypinhead2025, 2025-02-23:
Yes, exactly!

probonopd, 2025-02-24:
Hello, I lost track about this one a bit (I don't have the hardware for it). It sure looks great! Do we still need to do something in MiniDexed? Thanks.

craigyjp, 2025-02-24:
I bought a couple of these for another project to test but never got around to it. Maybe I will use one in future. I ended up with ST7789 displays for the colour.

[image]

zippypinhead2025, 2025-02-24:
SSD1306 OLEDs are too small. As a senior I need reading glasses for those tiny displays.
These SSD1305 hats are a good solution for a compact but readable device.

Sterr1, 2025-02-24:
> Do we still need to do something in MiniDexed? Thanks.
Not is really in MiniDexed. I download project, but my changes in circle not present. Update changes and compile project.

probonopd, 2025-02-24:
Right, so we will need to update to the latest Circle in MiniDexed.

Pending
* https://github.com/probonopd/MiniDexed/pull/755
* https://github.com/rsta2/circle/discussions/509

So it seems we can update Circle once it is at Step49.

Sterr1, 2025-02-28:
Today I received a display from China. As you can see, it’s smaller than my old one, and that one, in turn, is smaller than the standard 1602. If you remove the metal frame, there’s just a glass layer ~0.5-1 mm thick on a ribbon cable, which is visible on the other side of the board. So, you can glue the display to any glass from the inside and mount the board differently. On the board, you can see address selection via jumpers. But the addresses are listed incorrectly. The Chinese love to mislead buyers. The address is listed as 0x01111000b, but you just need to shift it right, and it becomes 0x00111100b = 0x3Ch. If you move the jumper, it will be 0x3Dh. The display worked immediately without any changes to the circuit or settings. But in old settings I was rotate to180 dg, and here it's reverted.  In China, the display costs 46 yuan + 2 yuan purchase tax on the Pinduoduo website. At today’s exchange rate, it’s ~$6.3. Shipping is very cheap, just slow. The same displays are available on Aliexpress, but they’re more expensive.

Also, in another photo, you can see an SD Card ribbon cable from China. The problem is that in the ZERO 2W, the card isn’t secured in any way and could fall out during an impact when mounted in a synthesizer. Plus, I didn’t want to mount the RPi with the connector facing outward close to the case wall. So, I chose an extension and a connector with a locking mechanism.

Once I solve all the issues with MIDI control (and for some reason, it works poorly in MiniDexed — maybe the MIDI adapter is bad), I’ll assemble everything into a MIDI keyboard. I’ll record a video and share it with you.

[image][image]

[image]

[image]

[image]

[image]

[image]

Sterr1, 2025-02-28:
> So it seems we can update Circle once it is at Step49.

I something don't understand on github. rsta2 make changes only in develop branch, but you use main branch, if I'm not mistaken. But in develop branch present two files ssd1306device.cpp, where changes applied, and ssd1306display.cpp where changes not applied. Files are similar.

diyelectromusic, 2025-02-28:
> But the addresses are listed incorrectly. The Chinese love to mislead buyers. The address is listed as 0x01111000b, but you just need to shift it right, and it becomes 0x00111100b = 0x3Ch.

They aren't misleading you, it is just a common problem with I2C addresses.

They are all 7-bit addresses, but it is the 7 most significant bits that are the address in the protocol as bit 0 is actually a R/W bit.  So if the 7-bit address is specified, then it is 0x3C, but if it is described as an 8-bit value then 0x3C with an additional R/W bit is indeed a shifted left address, in this case 0x78.

Kevin

probonopd, 2025-03-01:
> I something don't understand on github. rsta2 make changes only in develop branch, but you use main branch, if I'm not mistaken. But in develop branch present two files ssd1306device.cpp, where changes applied, and ssd1306display.cpp where changes not applied. Files are similar.

What is currently in the Circle develop branch will likely be in the next Circle release. At that point, we will update Circle and then support should be in MiniDexed.

probonopd, 2025-03-18:
Circle Step49 is out now, let's see how it performs at https://github.com/probonopd/MiniDexed/pull/755.

probonopd, 2025-04-18:
Is it working in the lastest MiniDexed build @Sterr1? Then we could close this. Thanks!

Sterr1, 2025-04-18:
Check strings was I added. If present, that work. I'm workout multisynth. Have not time to test it. Check my discuss about multisynth.

probonopd, 2025-04-18:
Please test it when you have time and let us know whether it works, because I don't have the hardware. Thanks!

Sterr1, 2025-04-18:
[image]

probonopd, 2025-04-18:
Thanks 👍 Anything else should be written in the Wiki?

* https://github.com/probonopd/MiniDexed/wiki/Hardware#ssd1305-i2c-oled

closed 2025-04-18 by probonopd

Sterr1, 2025-04-18:
No. All is OK! 👌