probonopd/MiniDexed #809: Minidexed Raspaudi -v3 hat no sound

Rotil66, 2025-02-08:
Hi Minidexed users,

I have a raspberry with minidexed that is working fine on the raspberry line line-out.
But on the Raspaudi -v3 hat that is recommended I get no sound.
Althought have enabled the 'SoundDevice=i2s' (and disabled the PWM)
What am I doing wrong or what must be setup differently to get is to work.
The link down here you can see that audiohat I have.

https://www.amazon.com/RASPIAUDIO-V3-Raspberry-Better-Quality/dp/B09T33LJQM?th=1

I hope somebody can help\me to get this to work.

Thanks,
Rolof

probonopd, 2025-02-08:
Hi @Rotil66, why not just use a GY-PCM5102 (less than 1/10th the price) and be done with it? I don't think there will be a difference in audio quality.

Rotil66, 2025-02-08:
I could if I had one,. This is the one I happen to have. In the tutorial video's they use this hat, so I'm wondering why this should not work ?? ( I agree, The difference in audio quality will not be noticable) and in the instruction they only say that I have to enable the I2s audio line. So it's a bit strange that is isn't working.
Any idea if I can try something to get this board working ??

Banana71, 2025-02-08:
Hi @Rotil66,
I have a Raspberry Pi 3 Model A+ running miniDexed with the RaspiAudio +V2 DAC without any issues. I don't know the differences between the two versions, but I believe the RaspiAudio +V3 also has the PCM5102 chip.

minidexed.ini

```
# Sound device
SoundDevice=i2s
#SoundDevice=pwm
#SoundDevice=hdmi
SampleRate=48000
ChunkSize=128
DACI2CAddress=0
ChannelsSwapped=0
# Engine Type ( 1=Modern ; 2=Mark I ; 3=OPL )
EngineType=1
```

[image]

Rotil66, 2025-02-08:
Hi Banana71,

I have the same settings as you have but no sound.
Can't be that different ??
By the way what does "Chunksize' do ??

Banana71, 2025-02-08:
The chunk size is the sample buffer and affects the latency. You can leave it commented out, then it corresponds to the value of 256.
Send us photos and your minidexed.ini, because now I have no clues as to why your configuration is failing. 
Please connect your miniDexed with HDMI to a monitor and take a screenshot after the boot process.

`#ChunkSize=256`

Rotil66, 2025-02-08:
#
# minidexed.ini
#

# Sound device
SoundDevice=i2s
#SoundDevice=pwm
#SoundDevice=hdmi
SampleRate=48000
ChunkSize=128
DACI2CAddress=0
ChannelsSwapped=0
# Engine Type ( 1=Modern ; 2=Mark I ; 3=OPL )
EngineType=1
#QuadDAC8Chan=0

# MIDI
MIDIBaudRate=31250
#MIDIThru=umidi1,ttyS1
IgnoreAllNotesOff=0
MIDIAutoVoiceDumpOnPC=0
HeaderlessSysExVoices=0
# Program Change enable
#   0 = Ignore all Program Change messages.
#   1 = Respond to Program Change messages.
MIDIRXProgramChange=1
# Program Change mode
#   0 = Only recognise Program Change 0-31.
#   1 = Support 0-127 across four consecutive banks.
# NB: Only relevant if PerformanceSelectChannel=0
ExpandPCAcrossBanks=1
# Program Change action:
#   0 = Program Change messages select voices on the channel associated with each TG.
#   1-16 = Program Change messages on this channel select performances.
#   >16 = Program Change messages on ANY channel select performances.
# NB: In performance mode, all Program Change messages on other channels are ignored.
PerformanceSelectChannel=0

# HD44780 LCD
LCDEnabled=1
LCDPinEnable=17
LCDPinRegisterSelect=4
LCDPinReadWrite=0
LCDPinData4=22
LCDPinData5=23
LCDPinData6=24
LCDPinData7=25
LCDI2CAddress=0x27

# SSD1306 LCD
# For a 128x32 display, set LCDColumns=20; LCDRows=2
# For a 128x64 display, set LCDColumns=20; LCDRows=4
SSD1306LCDI2CAddress=0x0
SSD1306LCDWidth=128
SSD1306LCDHeight=32
SSD1306LCDRotate=0
SSD1306LCDMirror=0

# ST7789 LCD
# SPIBus=0 for any RPi (GPIO 10,11,8,7).
#     Note: Leave blank (default) if no SPI device required.
# Select=0|1 for CE0 or CE1
# Data = GPIO pin number
# Optional: Reset, Backlight = GPIO pin numbers
# Rotation=0,90,180,270
# SmallFont=0 (default), 1
#
# For a 240 wide display set LCDColumns=15 with LCDRows=2
SPIBus=
ST7789Enabled=0
ST7789Data=
ST7789Select=
ST7789Reset=
ST7789Backlight=
ST7789Width=240
ST7789Height=240
ST7789Rotation=0
ST7789SmallFont=0

# Default is 16x2 display (e.g. HD44780)
LCDColumns=16
LCDRows=2

# GPIO Button Navigation
#  Any buttons set to 0 will be ignored
ButtonPinPrev=0
ButtonActionPrev=
ButtonPinNext=0
ButtonActionNext=
ButtonPinBack=11
ButtonActionBack=longpress
ButtonPinSelect=11
ButtonActionSelect=click
ButtonPinHome=11
ButtonActionHome=doubleclick
ButtonPinShortcut=11
# (Shortcut doesn't have an action)

# GPIO Program/Bank/TG Selection
#  Any buttons set to 0 will be ignored
ButtonPinPgmUp=13
ButtonActionPgmUp=click
ButtonPinPgmDown=0
ButtonActionPgmDown=
ButtonPinBankUp=26
ButtonActionBankUp=click
ButtonPinBankDown=19
ButtonActionBankDown=click
ButtonPinTGUp=0
ButtonActionTGUp=
ButtonPinTGDown=0
ButtonActionTGDown=

# Timeouts in milliseconds for double click and long press
DoubleClickTimeout=400
LongPressTimeout=400

# MIDI Button Navigation
#  Specify MIDI CC to act as a button (0 = ununsed, so don't use CC 0)
#  NB: Off < 64 < ON
# CC channel: 0=OFF; 1-16 MIDI Ch; >16 Omni
# If MIDIButtonNotes>0 then treat MIDIButton numbers as MIDI
# Note numbers, triggered with NoteOn/NoteOff, not CC numbers.
MIDIButtonCh=10
MIDIButtonNotes=0
# Arrow left
MIDIButtonPrev=96
# Arrow right
MIDIButtonNext=98
# Arrow up
MIDIButtonBack=100
# Arrow down
MIDIButtonSelect=102
# Home button
MIDIButtonHome=104
MIDIButtonPgmUp=106
MIDIButtonPgmDown=108
MIDIButtonBankUp=110
MIDIButtonBankDown=112
MIDIButtonTGUp=114
MIDIButtonTGDown=116

# KY-040 Rotary Encoder
EncoderEnabled=1
EncoderPinClock=10
EncoderPinData=9

# Debug
MIDIDumpEnabled=0
ProfileEnabled=0

# Performance
PerformanceSelectToLoad=1

Rotil66, 2025-02-08:
Sorry _ don'T know why the layout is this strange ??

Banana71, 2025-02-08:
Mark the Text from the minidexed.ini and press <>

[image]

Banana71, 2025-02-08:
What is the sound chip of your DAC labeled? PCM5102?

Rotil66, 2025-02-08:
It is marked: PCM5102A

Banana71, 2025-02-08:
But you probably plugged the jack plug into the audio out socket of the DAC? 
That's never happened to me before :-) , never :sunglasses:

Rotil66, 2025-02-08:
I do not understand this last remark (not native english)
(Where should it go otherwise ??)
I have plugged the jack in to the Audio out of the Dac Board, Raspberry Pi and tried the RCA connectors on the Dac board but none of them has audio.

Gustavomurta, 2025-02-08:
I recommend that you test your Audio +V3 module using the manufacturer's procedure. To make sure the module is working properly.

https://forum.raspiaudio.com/t/audio-installation-guide/19

Rotil66, 2025-02-08:
Ok I will. As soon as I'm back home

Rotil66, 2025-02-09:
Hi Gustavo,
Unfortunately I'm not able to perform this test because I'm not able to write full pi os to th sd. It always fails at verification amd the sd is empty. I manged to get a lite version to the SD.
It boot s but after some screen where I can enter credentials it ends in a cmd line prompt and there is where it end for me. Because I don't know how to get out of there.

I don't understand why writing a full os in the PI image software always fails. (no matter for what PI I write the image)

So I have no Idea how to perform this test.

Thanks

Gustavomurta, 2025-02-09:
Replace the SD card in the Rasp Pi. Use another SD card. It is not normal for an SD card to have writing problems with the SD imager.

Rotil66, 2025-02-09:
The sd card is OK. If I skip verifying, things run OK. This has always be the case using this PI imager. (on all my pc's)
Bu the Os is running OK and the soundboard is working fine. Also Minidexed is running fine there is only no sound thru this soundboard and that is weird. It is a PCM5102A and should not have a problem.

Rotil66, 2025-02-09:
Any idea how to get the board working. Since the board is technically OK??

Thanks!

Gustavomurta, 2025-02-09:
The PCM5102 chip that runs on the MiniDexed has some configured pins. If these pins on your module are not the same, it may be a problem.

https://github.com/dwhinham/mt32-pi/wiki/GY-PCM5102-DAC-module

Rotil66, 2025-02-09:
Hi Gustavo,
This board you show is not the board I use, I will check things. But Banana 71 is using the same (only version 2) board. It would be strange that the pinout would change from v2 to v3. But I try to find out.

Rotil66, 2025-02-11:
Hi All,

I have been checking but I'm pretty sure the board is OK since I can't find any difference to the V2. 
The board is also working perfectly with the MT32PI when set to I2s. 
So any Idea why it is not working with Minidexed ??

Thanks

soyersoyer, 2025-02-11:
> ButtonPinBankDown=19

Doesn't this conflict with the DAC?

Banana71, 2025-02-11:
Is the I2S connection displayed during booting?

Rotil66, 2025-02-11:
What should there be visible?

⁣BlueMail voor Android downloaden ​

Rotil66, 2025-02-11:
Have to check

⁣BlueMail voor Android downloaden ​

Rotil66, 2025-02-11:
Hi Soyersoyer,

That was probably the solution. I rewired the rotary to the pin-out as used for the Mt32pi. This was the solution The v3 hat is working now. 
(absolutely a great sound!!!)

Thank all for your help !!!!

Just another question.......
When I turn the rotary it takes several (unstable) clicks to get the next option/value. In the MT32Pi code there is an option where you can set this amount of clicks. Here every click in/decreases the value by 1. Is there an option for this in the minidexed software.
It's a bit 'unpleasant' when the amount of click is 'unstable' to get to the next value.

Thanks !!

soyersoyer, 2025-02-12:
Glad I could help,

Minidexed doesn't support it yet, but I'll find time to do it in 1-2 weeks.
https://github.com/probonopd/MiniDexed/discussions/803

Rotil66, 2025-02-12:
Hi all, it's great to see that this is a project that is very much 'alive' . Thanks a lot for that.
Is there somewhere a 'list' of what is being worked on? And is there something like a feature request for this.
( unfortunately I cannot code my self....)

I have seen that you can control the synth with  extra buttons and midi keys.
Bit what I would like to know if it's possible to have a button that directly opens the performance load screen. (and or a performance up/down button - and Performance bank up/down)
This would be nice for live playing

Thanks

Banana71, 2025-02-13:
I think a USB numeric keypad is a good idea for operating the menu. 
- Affordable for everyone 
- in combination with the KY-040 encoder, enough keys to conveniently operate the miniDexed
- Circle should support a keypad directly. 
- **_Only_** the UI of the miniDexed would have to be expanded/changed.

[image]

Rotil66, 2025-02-13:
Yes, that is also a good option. As long as interfacing get a bit more direct is it is great.
As long as the PI can handle it a bit more info the display (or support for bigger LCD) would be 'a nice to have'. But it's all volunteers that do  the job, so as long as I cannot code myself I cannot ask to much.
This allready a great project !!!
Thanks

Banana71, 2025-03-09:
Please close this post when the problems have been resolved

closed 2025-03-10 by probonopd