probonopd/MiniDexed #810: Cutoff and resonance

Sterr1, 2025-02-09:
When the resonance is set to 127, cutoff change from 0 to 20 is heard very gradually, with clicks. It's very non-analog sounding.

Banana71, 2025-03-09:
I cannot reproduce the error you describe. Which version of miniDexed are you using? Is the error still there?

probonopd, 2025-03-10:
Thanks for your report @Sterr1, Do you hear something different when you use [dexed](https://asb2m10.github.io/dexed/)on the PC? And in case you have access to a real hardware DX synthesizer, do you hear something different there? This would be very helpful to know.

Sterr1, 2025-03-10:
I disassembled my MiniDexed to mount it into a MIDI keyboard. So, at the moment, I can’t demonstrate this glitch. I’m slowly putting it together. When I’m done, I’ll record a video to make it clearer. PC version I don't use. 

> Which version of miniDexed are you using? Is the issue still there?

My kernel compiled from master with changes for SSD1305. I don't know what is version. You can try [this](https://github.com/probonopd/MiniDexed/issues/808#issuecomment-2676763390).

Sterr1, 2025-03-10:
I was try Dexed. The same. Preset "PHARON". [YOUTUBE](https://youtu.be/sxEYs139zes)

Sterr1, 2025-03-12:
I write issue to [Dexed](https://github.com/asb2m10/dexed/issues/471)