probonopd/MiniDexed #828: Noisy behavior during performance change and precision of rotary encoder

erresse2, 2025-03-26:
Hello,

First of all, thanks to all the people that made and still make this project so great and fun

I recently installed the latest version onto a pi zero 2W, with 16x2 LCD and classic rotatry encoder and PCM5102 I2S DAC...

And resulats are really great !!
At the moment, i am well satisfied with current functionnalities
Of couse, more would be great, but current featuers are already fantastic !

2 things could be improved or solved:
- when switching from a performance to another, i generally face a noise at the moment of the switch (this depends on the sound and possibly reverb)... I am not enough skilled to describe what happens, but it could be something like the DAC is not muted (despite it is set as per wiki instructions)... It is like, when you rotate the encoder by one notch and click, then a small noise appears...
it is much more evident when you still play the keayboard while switching to another performance...

- I noted that sometimes also, the precision of the rotatry encoder, when selecting another performance is not good... It looks like the LCD display is stucked while rotating the encoder... and suddenly the list of performance names start scrolling afetr a small delay... as if something was stucked..

The solving of these 2 issues (i cannot say it is bugs) would enhance this fun instrument... 

Regards

probonopd, 2025-04-20:
Hi @erresse2, in the future please open separate tickets for each issue.

Regarding issue 1: I think that what you hear is different levels of reverb. E.g., when you switch from a performance with little/short reverb to one with more/longer reverb, then you will hear a jump.

I wonder how hardware effects processors handle this.

Banana71, 2025-04-21:
Regarding the first issue:

I suspect, like both of you, that the reverb itself is the problem, and that some data in the buffer is still active after switching.
How I deal with this: If you don't want reverb in a performance, don't turn it off (Reverb Enable = off), but simply set the Reverb Level to 0.

Peter

Banana71, 2025-04-21:
Regarding the second issue:
@erresse2 Do you mean the problem with this discussion? #694 , https://github.com/probonopd/MiniDexed/discussions/694#discussioncomment-10653385

soyersoyer, 2025-04-23:
Does #883 fix the reverb problem?

erresse2, 2025-05-15:
Hello, sorry for not attending regularly this forum...  by the way, thanks for your answers !

i am a kind of newbie with github and not very familiar with the vocabulary of programers :-(   so it is quite difficult for me to search if a problem was already reported and already covered.... i noticed that new versions were now available, so i will test again and will revert...

well ...
about issue 2 : in fact i thougth reverb could be a reason for stucking microdexed ... i was more thinking that sometimes the change of program was not interrupting the sound generator or reverb generator... and minidexed was waiting for the end of the sound generation to change the preset... well, will try the new version and will revert.

about issue 1 : the small noise was also something quite subtile... hard to say if it was related to reverb or not... again, will try the new revision and will tell you...

erresse2, 2025-05-19:
hello, i tried the version of the build from May,02... and:
- still, when playing and switch from a performance to another , it makes like a scratch noise.. I think best would be to make a video of it...
- other point about freezing is much less visible in this version... I had more ease to change performances, it was much faster... but still sometime, it seems there is a small time lag.... again, the best would be to illustrate...

By the way, minidexed remains quite playable, this is for me just for better user experience.