probonopd/MiniDexed #846: Crackling sound

probonopd, 2025-04-19:
On RPi 3, this crackles in [0583421](https://github.com/probonopd/MiniDexed/commit/05834211383ae98e37c7a58f2ffec792e57d57af)

```
# Sound device
#SoundDevice=i2s
SoundDevice=pwm
#SoundDevice=hdmi
SampleRate=22000
#ChunkSize=256
DACI2CAddress=0x0
#DACI2CAddress=0x98
ChannelsSwapped=0
MasterVolume=32
```

The build in 
* https://github.com/probonopd/MiniDexed/pull/843
does not crackle.

We need to isolate the merge that caused the crackle to start.

probonopd, 2025-04-19:
6cbdab5f895a1810cd922d079a6c56d45a3fcf2b (Introduce MasterVolume, #842) also doesn't crackle.

So I suspect the crackle comes from #844, #845,

soyersoyer, 2025-04-19:
Could you try it without this commit?
https://github.com/probonopd/MiniDexed/pull/844/commits/c918b5985213340ed8b43e598fbf7fe378333e35

probonopd, 2025-04-19:
For now I have restored main to a non-crackling state. Let's add back changes slowly, testing each change with the config above. Thanks!

(I am right now working on merging the networking stuff and was surprised to hear crackle even though networking was not even switched on ;-))

Banana71, 2025-04-19:
Just curious.
Why are you using such a strange sample rate of 22000?
A common sample rate in that range would be 22050.

In direct comparison, 48000 sounds much better. Doesn't that work for me? Or why don't you set it to 48000?

Banana71, 2025-04-19:
I also wanted to mention here again, just like in issue #846, that my Raspberry Pi 4 crashes with a configured sample rate of 22000.

soyersoyer, 2025-04-19:
I don't notice any difference in the PWM output of my raspberry pi 3b+. Just as noisy. How is it crackling for you?

probonopd, 2025-04-19:
@soyersoyer have you used the settings I pasted above? Maybe something to do with the sample rate and/or in combination with the Master Volume? For me it crackled really bad to  the point of completely being unusable.

soyersoyer, 2025-04-19:
Yeah, I tried that too. It's quite strange.

probonopd, 2025-04-19:
So for you it is not crackling?

soyersoyer, 2025-04-19:
With my 3b+, no.

probonopd, 2025-04-19:
Thanks for your individual pull requests. It is strange - so far I don't get the crackle. anywhere..

closed 2025-05-19 by probonopd