probonopd/MiniDexed #847: HDMI crashes

probonopd, 2025-04-19:
On RPi 3, HDMI crashes during boot, in the main branch as of [0583421](https://github.com/probonopd/MiniDexed/commit/05834211383ae98e37c7a58f2ffec792e57d57af)

```
# Sound device
#SoundDevice=i2s
#SoundDevice=pwm
SoundDevice=hdmi
SampleRate=22000
#ChunkSize=256
DACI2CAddress=0x0
#DACI2CAddress=0x98
ChannelsSwapped=0
MasterVolume=32
```

Banana71, 2025-04-19:
I don't think it's exclusively an HDMI issue. My Raspberry Pi 4 with an I2S DAC also crashes at a sample rate of 22050, even with older miniDexed versions.

Banana71, 2025-04-19:
Does the HDMI audio output work with a sample rate of 48000Hz?

probonopd, 2025-04-19:
With `SampleRate=44800` it also crashes in `soundbasedevice.cpp`.

probonopd, 2025-04-19:
With `SampleRate` entirely commented out, it doesn't show the crash message on HDMI but freezes after `CPU core 2 started`.

soyersoyer, 2025-04-19:
Do you mean `SampleRate=44100`?

probonopd, 2025-04-19:
My bad, I have not used one of the "common" sampling rates; will need to retest with those.

Ideally, we make this hardcoded and not configurable. One variable less to worry about.
Or at least accept only the ones posted above.

soyersoyer, 2025-04-19:
Reverting Commit f546448 ("Update circle to Step49") the HDMI works.

probonopd, 2025-04-19:
@Banana71 48000 also crashes.

probonopd, 2025-04-19:
> Reverting Commit https://github.com/probonopd/MiniDexed/commit/f546448371c4d630fff2d66ac2083711dc0e63bc ("Update circle to Step49") the HDMI works.

Wow, how did you find the culprit so quickly?

Does this look like it needs to be fixed in Circle?

soyersoyer, 2025-04-19:
I thought this was the 3rd most likely. Yes, I think the problem is in Circle.

soyersoyer, 2025-04-19:
This is the first bad commit within Circle:
https://github.com/rsta2/circle/commit/f986fa501c33ed45d98bf7ff9af0e7762843de58

probonopd, 2025-04-19:
@rsta2 trying to use HDMI audio hangs MiniDexed (apparently since Step49), is there a good/easy way to verifiy whether HDMI audio still works in Circle? Or maybe you already have a clue as to what this might be causing this?

probonopd, 2025-04-28:
@soyersoyer 

> Could it be that the change involved was not implemented in CHDMISoundBaseDevice?

I do not understand the meaning of this.

The only mention of `CHDMISoundBaseDevice` in MiniDexed is

```
		m_pSoundDevice = new CHDMISoundBaseDevice (pInterrupt, pConfig->GetSampleRate (),
							   pConfig->GetChunkSize ());
```

soyersoyer, 2025-04-28:
After the mentioned commit, the CDMAChannel's SetCompletionRoutine "Must be called before starting every new asynchronous transfer." CHDMISoundBaseDevice uses the CDMAChannel, and the SetCompletionRoutine has only been called once. I think the CHDMISoundBaseDevice::DMACompletionRoutine should call the SetCompletionRoutine somewhere, but I've never tried.