probonopd/MiniDexed #856: Show name when saving performance

Octet-nl, 2025-04-20:
Hello,

When I want to save a performance, only the text "Overwrite" is shown. It would be very handy if the name of the performance is also shown.  Now I can only hope that I selected the right performance. 

[image]

Also when the save is completed no name is visible.

[image]

A confirmation is nice, but a confirmation of the overwritten name would be more helpfull.

Kind regards,

Hans.

soyersoyer, 2025-04-20:
`<ot>`
This design looks cool! Is it available somewhere?
`</ot>`

Octet-nl, 2025-04-20:
`<ot>`

Do you mean the casing? My wife thinks it's bizarre and says it could have come straight out of Doctor Who.

[image]

The issue tracker is not the right place, but if you’d like, I can open a discussion and describe it there.

`</ot>`

soyersoyer, 2025-04-20:
`<ot>`
Sure, I like it! You can't click with the wheel, but there is an OK button, right?

If someone asked you what it is, and you would say it's a time machine, it would be believable. :)
`</ot>`

probonopd, 2025-04-20:
Almost like a [Rolodex](https://en.wikipedia.org/wiki/Rolodex). RoloDexed ;-)

Octet-nl, 2025-04-20:
`<ot>`
You can still click the wheel, it's the HOME function.
See description:
https://git0hub.com/probonopd/MiniDexed/discussions/857
`</ot>`

Octet-nl, 2025-04-20:
I hope people don't get too much distracted from the original question ;)

probonopd, 2025-04-20:
Well, well, well... the display has only so much space and there is an overall concept for the menus which mandates what goes into the top line. So instead of the word "Overwrite" we could show the name of the performance. Would that be an improvement?

Octet-nl, 2025-04-20:
Certainly. 
The problem arises when selecting a performance, then make a lot of adaptations to voices, channels etc.
Then, after a lot of fiddling I want to save the performance, but I get no clue if it is still the correct one.
Also, if I forget to select a performance and afterwards want to save changes, I don't have any visual feedback that I'm making a mistake. To be certain, I now often select New, and afterwards rename the performance on the SD card.
Any visual feedback is welcome. An alternative would be to insert an extra confirmation step, 
"performance_name are you sure?
But that would be an extra menu level and I can imagine that is a lot more work (and more than it's worth).
So, if the word "Overwrite" is replaced by the name of the performance that would definitely be an improvement.

Banana71, 2025-04-20:
I understand your problem. I've always managed to get by by going to the Performance - Load menu. The [L] is displayed next to the currently loaded Performance. This way you can be sure to overwrite the correct Performance.

probonopd, 2025-04-20:
Upon closer inspection, I think the behavior for overwriting should like the existing behavior for deleting.

Octet-nl, 2025-04-20:
Wow! That would be even better! 
Just select the name you want to overwrite.
I would be very happy with that (and I suppose Banana71 too :-)