probonopd/MiniDexed #872: Dexed can't switch on/off or control the OPs in the MiniDexed

probonopd, 2025-04-22:
* Dexed **can** control Transpose in MiniDexed
* Dexed **can't** control Detune in MiniDexed
* Dexed **can't** switch on/off or control the OPs in the MiniDexed - why is this?

Is there a bug that prevents per-operator MIDI commands from being executed?

probonopd, 2025-04-23:
Dexed sends to switch operators ON when initially all are off:

```
0xf0
0x43
0x10
0x01
0x1b
0x20 for OP1, 0x30 for OP2, 0x38 for OP3, 0x3c for OP4, 0x3e for OP5, 0x3f for OP6
0xf7
```

Dexed sends to switch operators Off when initially all are on:

```
0xf0
0x43
0x10
0x01
0x1b
0x1f for OP1, 0x0f for OP2, 0x07 for OP3, 0x03 for OP4, 0x01 for OP5, 0x00 for OP6
0xf7
```

So in other words, what gets sent in byte 6 is a combination of the state of all operators.

```
/*
- The sixth byte sent by Dexed is a packed bit-field representing the on/off states of six operators.
- Each operator corresponds to a specific bit in the sixth byte:
    Bit 5: Operator 1
    Bit 4: Operator 2
    Bit 3: Operator 3
    Bit 2: Operator 4
    Bit 1: Operator 5
    Bit 0: Operator 6
- To determine an operator’s state, we check if its corresponding bit is set (1 for ON, 0 for OFF).
- This encoding allows a single byte to represent the state of all six operators compactly.
*/
```

Skerjanc, 2025-04-27:
> - Each operator corresponds to a specific bit in the sixth byte:
>     Bit 5: Operator 1
>     Bit 4: Operator 2
>     Bit 3: Operator 3
>     Bit 2: Operator 4
>     Bit 1: Operator 5
>     Bit 0: Operator 6

Yes, I confirm this is right!

probonopd, 2025-04-28:
Disabling all operators in Dexed on the PC results in MiniDexed logging

```
00:17:12.427 >1 - 192.168.0.59 mididevice - - - MIDI24: SysEx data length: [7]:
00:17:12.428 >1 - 192.168.0.59 mididevice - - - 0000:
00:17:12.428 >1 - 192.168.0.59 mididevice - - -  0xf0
00:17:12.428 >1 - 192.168.0.59 mididevice - - -  0x43
00:17:12.430 >1 - 192.168.0.59 mididevice - - -  0x10
00:17:12.430 >1 - 192.168.0.59 mididevice - - -  0x01
00:17:12.430 >1 - 192.168.0.59 mididevice - - -  0x1b
00:17:12.431 >1 - 192.168.0.59 mididevice - - -  0x00
00:17:12.431 >1 - 192.168.0.59 mididevice - - -  0xf7
00:17:12.432 >1 - 192.168.0.59 mididevice - - -
00:17:12.432 >1 - 192.168.0.59 mididevice - - - MIDI-SYSEX: channel: 0, len: 7, TG: 0
00:17:12.433 >1 - 192.168.0.59 mididevice - - - SYSEX handler return value: 455
00:17:12.434 >1 - 192.168.0.59 mididevice - - - SysEx voice parameter change: Parameter 155 value: 0
```

In the VCED format, parameter 155 is indeed operator enable/disable (OPE) as per Yamaha documentation:

[image]

But newly triggered notes are not silent . This probably means that the operator enable mask is being set to 0 by MiniDexed, but the Synth_Dexed synth engine is not actually disabling the operators. So I need to check Synth_Dexed to see what it does with parameter 155.

PR #896 (2025-04-28): Fix operator enable/disable
Closes #872
commit f03e3887 (2025-04-28): Fix operator enable/disable (#896)

Enable/disable operators, closes #872
by calling m_pSynthesizer->setOPMask
closed 2025-04-28 by probonopd