probonopd/MiniDexed #875: Getting lots of "icmp - - - Time exceeded (0)" from circle-stdlib

probonopd, 2025-04-22:
Getting lots of "icmp - - - Time exceeded (0)" even though the networking functionality is working.

The message is coming from `circle-stdlib/libs/circle/lib/net/icmphandler.cpp`, so it is not really under our control.

We could patch it away on-the-fly with `sed -i "/case ICMP_TYPE_TIME_EXCEED:/,/break;/d" circle-stdlib/libs/circle/lib/net/icmphandler.cpp` but that doesn't feel like a clean solution.

diyelectromusic, 2025-04-22:
Please don't patch code on the fly :)

Is this related to https://github.com/probonopd/MiniDexed/pull/747#issuecomment-2481434797

Kevin

probonopd, 2025-04-22:
Yes, it is not only related but it is the exact same (imho cosmetic) message. Because everything works fine. Possibly the message is appearing without there being an actual error?

probonopd, 2025-04-23:
What is it trying to achieve with icmp in the first place? Do we need that?

diyelectromusic, 2025-04-23:
icmp is pretty critical in IP networks. If it is reporting issues we ought to look into why. Icmp handles all sorts of IP level control information about routing and passing of packets (the clues in the name - Internet Control Message Protocol).

Of course it could be bugs in our use of the network stack, or even bugs in the stack itself - I don't know how well used in circle it has been.

But it could signify that we have some of the network parameters wrong - maybe window sizes/buffers too small; network related timeouts too short; etc.

It could be that networking is functioning but with lots of retries or timeouts, therefore loading the system or the network (or both), but it still manages to get packets through.

It could be genuine issues on the network of course.

Or it could be that we're not servicing the networking code often enough so it isn't getting to handle packets in time to satisfy the network stack (see my other comment about the difficulties of networks on embedded and real-time systems).

I short, I strongly suggest we don't ignore it, but instead try to understand why it is happening.

But I am not familiar with the circle networking stack at all I'm afraid, so if would take me a lot of effort, or someone else who is more knowledgeable about it should have a look.

Perhaps the original author would like to step up to help maintain the code they contributed?

Kevin

probonopd, 2025-04-23:
Maybe @rsta2 has an idea why we are getting flooded by the `icmp - - - Time exceeded (0)` message? Thanks.

probonopd, 2025-12-23:
Haven't tried lately, but I used to see this when using syslog (as described in the wiki). Don't have my hardware at hand to test at the moment. Are you no longer seeing it? Maybe the router also plays a role in this?

probonopd, 2026-11-15:
I traced where this message comes from on current `main` (Circle commit as pinned by `circle-stdlib`).

`libs/circle/lib/net/icmphandler.cpp`, in the error-message branch, handles `ICMP_TYPE_TIME_EXCEED` and writes:

`CLogger::Get()->Write(FromICMP, LogWarning, "Time exceeded (%u)", pICMPHeader->nCode);`

So this is Circle reporting an actual ICMP Time Exceeded packet received by the Pi, not a MiniDexed-generated log. It can come from a router/firewall, a stale or wrong syslog destination, or generally from a packet that expired in transit. It is not something MiniDexed can filter per-module today: there is no app-level logger filter for Circle's ICMP module, and suppressing `LogWarning` globally would hide unrelated warnings.

The comments above (syslog being the trigger, destination unreachable when the syslog server is down, router involvement) fit that explanation. Patching the Circle source just to remove the message would hide useful network information and is better handled upstream if it is considered too noisy.

Suggested next step before treating this as a MiniDexed TODO: reproduce with a known configuration and report:
- whether it disappears with `NetworkSyslogEnabled=0` and/or with a known-good syslog destination
- the network setup and whether the messages correlate with syslog/UDP traffic
- a packet capture showing the ICMP type/code and the packet that triggered it

If it turns out to be a Circle behaviour issue, it should be filed/patched in `rsta2/circle` rather than adding a workaround in MiniDexed.