probonopd/MiniDexed #881: Sending cartridges over MIDI to MiniDexed fails

probonopd, 2025-04-22:
Sending cartridges over MIDI to MiniDexed fails, e.g., using https://bladesk.github.io/DX7II-Librarian/

We get

```
00:55:50.987 >1 - 192.168.0.59 mididevice - - - MIDI-SYSEX: channel: 0, len: 255, TG: 0
00:55:50.989 >1 - 192.168.0.59 mididevice - - - SYSEX handler return value: -1
00:55:50.989 >1 - 192.168.0.59 mididevice - - - SysEx end status byte not detected.
00:55:50.989 >1 - 192.168.0.59 mididevice - - - MIDI-SYSEX: channel: 0, len: 255, TG: 1
00:55:50.989 >1 - 192.168.0.59 mididevice - - - SYSEX handler return value: -1
00:55:50.990 >1 - 192.168.0.59 mididevice - - - SysEx end status byte not detected.
00:55:50.991 >1 - 192.168.0.59 mididevice - - - MIDI-SYSEX: channel: 0, len: 255, TG: 2
00:55:50.991 >1 - 192.168.0.59 mididevice - - - SYSEX handler return value: -1
00:55:50.991 >1 - 192.168.0.59 mididevice - - - SysEx end status byte not detected.
00:55:50.991 >1 - 192.168.0.59 mididevice - - - MIDI-SYSEX: channel: 0, len: 255, TG: 3
00:55:50.991 >1 - 192.168.0.59 mididevice - - - SYSEX handler return value: -1
00:55:50.991 >1 - 192.168.0.59 mididevice - - - SysEx end status byte not detected.
00:55:50.991 >1 - 192.168.0.59 mididevice - - - MIDI-SYSEX: channel: 0, len: 255, TG: 4
00:55:50.992 >1 - 192.168.0.59 mididevice - - - SYSEX handler return value: -1
00:55:50.992 >1 - 192.168.0.59 mididevice - - - SysEx end status byte not detected.
00:55:50.992 >1 - 192.168.0.59 mididevice - - - MIDI-SYSEX: channel: 0, len: 255, TG: 5
00:55:50.992 >1 - 192.168.0.59 mididevice - - - SYSEX handler return value: -1
00:55:50.992 >1 - 192.168.0.59 mididevice - - - SysEx end status byte not detected.
```

(Gosh, that syslog via network is awesome - no more taking pictures of HDMI screens!)

probonopd, 2025-04-22:
Looking at

https://codeberg.org/dcoredump/Synth_Dexed/src/commit/d2e73d3ab9e958989342032ac7781d198183efb0/src/dexed.cpp#L975-L977

```
int16_t Dexed::checkSystemExclusive(const uint8_t* sysex, const uint16_t len)
/*
        -1:     SysEx end status byte not detected.
        -2:     SysEx vendor not Yamaha.
        -3:     Unknown SysEx parameter change.
        -4:     Unknown SysEx voice or function.
        -5:     Not a SysEx voice bulk upload.
        -6:     Wrong length for SysEx voice bulk upload (not 155).
        -7:     Checksum error for one voice.
        -8:     Not a SysEx bank bulk upload.
        -9:     Wrong length for SysEx bank bulk upload (not 4096).
        -10:    Checksum error for bank.
        -11:    Unknown SysEx message.
	64-77:	Function parameter changed.
	100:	Voice loaded.
	200:	Bank loaded.
	300-455:	Voice parameter changed.
*/
{
  int32_t bulk_checksum_calc = 0;
  const int8_t bulk_checksum = sysex[161];

  // Check for SYSEX end byte
  if (sysex[len - 1] != 0xf7)
    return(-1);
```

this seems to suggest that we don't feed the entire `sysex` in, and/or give it a wrong `len`?

BobanSpasic, 2025-04-29:
MiniDexed accepts just VCED (one voice) dumps. VMEM (32-voice) dumps aren't processed in any way. There was a discussion with @dcoredump about this long time ago. There is no clear way of doing it like on a real DX7. One way of doing this is to write the dump directly to voice directory, and ask the user for a file name after the dump finishes.

probonopd, 2025-04-29:
> VMEM (32-voice) dumps aren't processed in any way. 

Yes, that's the "bug" ;-)

probonopd, 2025-05-01:
Thinking about it... you definitely have a point :-)

The behavior I intend for MiniDexed is that the bank that is currently selected for the TG(s) that listens on the MIDI channel gets overwritten (in RAM; MiniDexed doesn't write .syx files back anyhow yet).

So maybe this should be handled entirely outside of Synth_Dexed, would you agree?

maxcomperatore, 2026-09-13:
The log lines tell the exact story:
```
MIDI-SYSEX: channel: 0, len: 255, TG: 0
SYSEX handler return value: -1
SysEx end status byte not detected.
```

A standard Yamaha DX7 32-voice bulk cartridge dump is exactly **4,104 bytes** (`F0 43 00 09 20 00` + 4,096 bytes data + checksum + `F7`).

What is happening here:
1. The incoming MIDI driver/buffer is capped at a fixed single-buffer length of 255 bytes (or 256 bytes). When the buffer fills up without seeing the terminal `0xF7` byte, the handler bails with return value `-1` ("SysEx end status byte not detected").
2. Furthermore, the log shows the partial 255-byte chunk is being dispatched across Tone Generators `TG: 0` through `TG: 5` in an uncoordinated broadcast instead of accumulating into a unified SysEx assembly buffer.

To receive DX7 bulk dumps, the MIDI stack needs an accumulation state machine: when an `0xF0` arrives, append chunks until `0xF7` is received or an error/timeout occurs, with a buffer sized for at least 4,104 bytes. If memory is tight on bare metal Raspberry Pi, streaming the 128-byte voice records directly into patch storage as they pass through is an option, but checking the trailing 1-byte 7-bit two's complement checksum requires keeping the rolling sum of all 4,096 bytes.

We hit this exact 4,104-byte bulk framing constraint when building the Web MIDI cartridge manager at [bipluk.com](https://bipluk.com). Full 32-voice cartridges will always overflow any sub-4KB buffer.