probonopd/MiniDexed #882: Requesting cartridge dumps over MIDI from MiniDexed fails

probonopd, 2025-04-22:
Requesting cartridge dumps over MIDI from MiniDexed fails, e.g., using https://bladesk.github.io/DX7II-Librarian/, top-right hamburger menu, "Request voices from a DX7".

We get

```
00:58:51.312 >1 - 192.168.0.59 mididevice - - - MIDI-SYSEX: channel: 0, len: 5, TG: 0
00:58:51.312 >1 - 192.168.0.59 mididevice - - - SYSEX handler return value: -11
00:58:51.312 >1 - 192.168.0.59 mididevice - - - Unknown SysEx message.
00:58:51.312 >1 - 192.168.0.59 mididevice - - - MIDI-SYSEX: channel: 0, len: 5, TG: 1
00:58:51.312 >1 - 192.168.0.59 mididevice - - - SYSEX handler return value: -11
00:58:51.312 >1 - 192.168.0.59 mididevice - - - Unknown SysEx message.
00:58:51.313 >1 - 192.168.0.59 mididevice - - - MIDI-SYSEX: channel: 0, len: 5, TG: 2
00:58:51.313 >1 - 192.168.0.59 mididevice - - - SYSEX handler return value: -11
00:58:51.313 >1 - 192.168.0.59 mididevice - - - Unknown SysEx message.
00:58:51.313 >1 - 192.168.0.59 mididevice - - - MIDI-SYSEX: channel: 0, len: 5, TG: 3
00:58:51.314 >1 - 192.168.0.59 mididevice - - - SYSEX handler return value: -11
00:58:51.314 >1 - 192.168.0.59 mididevice - - - Unknown SysEx message.
00:58:51.314 >1 - 192.168.0.59 mididevice - - - MIDI-SYSEX: channel: 0, len: 5, TG: 4
00:58:51.314 >1 - 192.168.0.59 mididevice - - - SYSEX handler return value: -11
00:58:51.314 >1 - 192.168.0.59 mididevice - - - Unknown SysEx message.
00:58:51.314 >1 - 192.168.0.59 mididevice - - - MIDI-SYSEX: channel: 0, len: 5, TG: 5
00:58:51.314 >1 - 192.168.0.59 mididevice - - - SYSEX handler return value: -11
00:58:51.315 >1 - 192.168.0.59 mididevice - - - Unknown SysEx message.
```

(Gosh, that syslog via network is awesome - no more taking pictures of HDMI screens!)

probonopd, 2025-04-23:
Looking at

https://codeberg.org/dcoredump/Synth_Dexed/src/commit/d2e73d3ab9e958989342032ac7781d198183efb0/src/dexed.cpp#L975-L977

```
int16_t Dexed::checkSystemExclusive(const uint8_t* sysex, const uint16_t len)
/*
(...)
        -11:    Unknown SysEx message.
```

It seems like responding to cartridge dump requests over MIDI would need to be implemented in Synth_Dexed.

probonopd, 2025-04-23:
Related discussion:
* https://github.com/probonopd/MiniDexed/pull/837 (the part on voice dumps)

It seems like responding to voice dump requests over MIDI would also ideally be implemented in Synth_Dexed.

probonopd, 2025-04-23:
Voice Dump Request via SysEx: MiniDexed should listen for the specific SysEx message (`F0 43 20 00 F7`) and send the voice dump for the requested channel. (Reference: https://github.com/asb2m10/dexed/issues/141) - Possibly should be handled by Synth_Dexed?

Skerjanc, 2025-04-27:
> Voice Dump Request via SysEx: MiniDexed should listen for the specific SysEx message (`F0 43 20 00 F7`) and send the voice dump for the requested channel. (Reference: [asb2m10/dexed#141](https://github.com/asb2m10/dexed/issues/141)) - Possibly should be handled by Synth_Dexed?

Also 32 voice bulk data dump shall be possible after the bulk request ('f0 43 20 09 f7 ')
If needed, I could share the code to unpack the bulk dump.

probonopd, 2025-04-28:
Relevant documentation: https://github.com/asb2m10/dexed/blob/master/Documentation/sysex-format.txt

```
////////////////////////////////////////////////////////////
A:
SYSEX Message: Bulk Data for 1 Voice
------------------------------------
       bits    hex  description

     11110000  F0   Status byte - start sysex
     0iiiiiii  43   ID # (i=67; Yamaha)
     0sssnnnn  00   Sub-status (s=0) & channel number (n=0; ch 1)
     0fffffff  00   format number (f=0; 1 voice)
     0bbbbbbb  01   byte count MS byte
     0bbbbbbb  1B   byte count LS byte (b=155; 1 voice)
     0ddddddd  **   data byte 1

        |       |       |

     0ddddddd  **   data byte 155
     0eeeeeee  **   checksum (masked 2's complement of sum of 155 bytes)
     11110111  F7   Status - end sysex

///////////////////////////////////////////////////////////
B:
SYSEX Message: Bulk Data for 32 Voices
--------------------------------------
       bits    hex  description

     11110000  F0   Status byte - start sysex
     0iiiiiii  43   ID # (i=67; Yamaha)
     0sssnnnn  00   Sub-status (s=0) & channel number (n=0; ch 1)
     0fffffff  09   format number (f=9; 32 voices)
     0bbbbbbb  20   byte count MS byte
     0bbbbbbb  00   byte count LS byte (b=4096; 32 voices)
     0ddddddd  **   data byte 1

        |       |       |

     0ddddddd  **   data byte 4096  (there are 128 bytes / voice)
     0eeeeeee  **   checksum (masked 2's comp. of sum of 4096 bytes)
     11110111  F7   Status - end sysex

///////////////////////////////////////////////////////////////

F:
Data Structure: Bulk Dump Packed Format
---------------------------------------

OK, now the tricky bit. For a bulk dump the 155 voice parameters for each
 voice are packed into 32 consecutive 128 byte chunks as follows ...

byte             bit #
 #     6   5   4   3   2   1   0   param A       range  param B       range
----  --- --- --- --- --- --- ---  ------------  -----  ------------  -----
  0                R1              OP6 EG R1      0-99
  1                R2              OP6 EG R2      0-99
  2                R3              OP6 EG R3      0-99
  3                R4              OP6 EG R4      0-99
  4                L1              OP6 EG L1      0-99
  5                L2              OP6 EG L2      0-99
  6                L3              OP6 EG L3      0-99
  7                L4              OP6 EG L4      0-99
  8                BP              LEV SCL BRK PT 0-99
  9                LD              SCL LEFT DEPTH 0-99
 10                RD              SCL RGHT DEPTH 0-99
 11    0   0   0 |  RC   |   LC  | SCL LEFT CURVE 0-3   SCL RGHT CURVE 0-3
 12  |      DET      |     RS    | OSC DETUNE     0-14  OSC RATE SCALE 0-7
 13    0   0 |    KVS    |  AMS  | KEY VEL SENS   0-7   AMP MOD SENS   0-3
 14                OL              OP6 OUTPUT LEV 0-99
 15    0 |         FC        | M | FREQ COARSE    0-31  OSC MODE       0-1
 16                FF              FREQ FINE      0-99
 17 \
  |  > these 17 bytes for OSC 5
 33 /
 34 \
  |  > these 17 bytes for OSC 4
 50 /
 51 \
  |  > these 17 bytes for OSC 3
 67 /
 68 \
  |  > these 17 bytes for OSC 2
 84 /
 85 \
  |  > these 17 bytes for OSC 1
101 /

byte             bit #
 #     6   5   4   3   2   1   0   param A       range  param B       range
----  --- --- --- --- --- --- ---  ------------  -----  ------------  -----
102               PR1              PITCH EG R1   0-99
103               PR2              PITCH EG R2   0-99
104               PR3              PITCH EG R3   0-99
105               PR4              PITCH EG R4   0-99
106               PL1              PITCH EG L1   0-99
107               PL2              PITCH EG L2   0-99
108               PL3              PITCH EG L3   0-99
109               PL4              PITCH EG L4   0-99
110    0   0 |        ALG        | ALGORITHM     0-31
111    0   0   0 |OKS|    FB     | OSC KEY SYNC  0-1    FEEDBACK      0-7
112               LFS              LFO SPEED     0-99
113               LFD              LFO DELAY     0-99
114               LPMD             LF PT MOD DEP 0-99
115               LAMD             LF AM MOD DEP 0-99
116  |  LPMS |      LFW      |LKS| LF PT MOD SNS 0-7   WAVE 0-5,  SYNC 0-1
117              TRNSP             TRANSPOSE     0-48
118          NAME CHAR 1           VOICE NAME 1  ASCII
119          NAME CHAR 2           VOICE NAME 2  ASCII
120          NAME CHAR 3           VOICE NAME 3  ASCII
121          NAME CHAR 4           VOICE NAME 4  ASCII
122          NAME CHAR 5           VOICE NAME 5  ASCII
123          NAME CHAR 6           VOICE NAME 6  ASCII
124          NAME CHAR 7           VOICE NAME 7  ASCII
125          NAME CHAR 8           VOICE NAME 8  ASCII
126          NAME CHAR 9           VOICE NAME 9  ASCII
127          NAME CHAR 10          VOICE NAME 10 ASCII

/////////////////////////////////////////////////////////////////////
```

Now the question, which format (A? B? F? Something else?) does MiniDexed need to send when it receives

* `F0 43 20 00 F7` --> Format A?
* `F0 43 20 09 F7` --> Format B?

> If needed, I could share the code to pack the bulk dump.

Do you mean format F by that? When would MiniDexed need to send that?

Skerjanc, 2025-04-29:
> * `F0 43 20 00 F7` --> Format A? F=0, yes, this is **One Voice Bulk Data**, 155 bytes.
> * `F0 43 20 09 F7` --> Format B? F=9, yes **32 Voice Bulk Data**, 32*128 bytes , or F format from above. The parameters must be reordered, some with less bits are packed together.
3rd byte of the dump request must be (0x20 | MIDIchannel), bit 6 is the actual request.

Only the module corresponding to the MIDI channel (alias modul number) is to be dumped per request.

probonopd, 2025-04-29:
By the way, the thoughts above not only concern cartridge dumps but basically all SysEx.

probonopd, 2025-04-29:
I think we want to implement TX816 behavior. Thus I need to understand how the TX816 behaves :-)

BobanSpasic, 2025-04-29:
Let's understand the architecture of a DX7:
- there is something called Edit buffer, containing one current voice. A VCED dump over MIDI goes here.
- there is an Edit buffer for functions (a set of parameters under the name "functions")
- there is a bank of 32 voices in memory (VMEM dump goes here). At selecting a voice, it is copied to voice edit buffer, and now you can play or edit it.
- there are global settings called functions on DX7, and these are equivalent to the setting for one TG in our performances (portamento, MW, BC, FC, AT etc settings). That was the biggest shortcoming of the DX7 - just one global set for the parameters called "functions".

TX7:
- expander for a DX7. It was constructed not as a stand-alone, but to be connected to a DX7. It can also work as a stand-alone, but the extra capabilities described in the next line will remain unused.
- a TX7 bank contains 32 voices and 64 functions. At selecting a voice 1 on TX7, it will load voice 1 and function 1, and it will select voice 1 on a connected DX7, plus it will send the function 33 to the DX7. This way, it overcomes the shortcoming of the DX7 of having just one global function set. The global function set on DX7 will be overwritten every time at selecting a voice on TX7.
- with functions, you can now have a dual voice or a split, detuned dual etc.

TX812:
- 2x stand-alone TX7 in one device - two sound generators with functions per voice
- functions are just like on TX7 - a set of 64 functions, where just the first 32 are used, linked each to one voice from the bank - at loading a voice, the corresponding functions set is also loaded.

TX816:
- 8x stand-alone TX7

DX7s:
- a "better" DX7 - 16bit (instead of 12bit), some better MIDI implementation, but still DX7 compatible.

DX7II (new generation that in some settings breaks the compatibility with the above devices):
- a voice bank for DX7II contains voice data (more or less DX7 compatible) + supplement for each voice in the bank. Supplement are additional parameters compared to the original DX7. Some DX7 function parameters are moved here (that what makes sense to be set per voice, like the settings for MW, FC, BC, Portamento etc.)
- Additional layer is called Performance, and it combines 2 voices (incl. supplement) plus additional parameters like keyboard range, detune etc. So, some of the TX7/TX816 functions are located here.
- it can load a DX7 voice banks, but some voices will not sound the same (manual tuning needed). It is about the range of some of the parameters, where for example (fictional, as I am too lazy to look for a real example) one parameter on DX7 has a range 0-7, but it is 0-15 on DX7II (better precision).

TX802:
- 4x DX7II = 8 sound engines
- panning effects (you can define the panning envelope how the sound will move through the channels)
- "link" function - you can link a couple of engines, and every incoming note will be played on the next engine in the chain (in the link). It was meant for overcoming the polyphony-shortage, but you can get some nice effects if the linked engines have different voices loaded.

This would be some coarse description of the architecture differences.

probonopd, 2025-04-29:
So, what do you predict will be the outcome of "Test on TX816" above? Honestly I don't know what the outcome will be.

BobanSpasic, 2025-04-29:
1. you can either send single-voice dump request or bulk dump request (32 voice bank). Not the same SysEx message.
2. if you send single-voice dump request, you'll get 163 bytes from the edit buffer (currently loaded voice). Headers are included, not just data
3. if you send bulk dump request (32 voices), you'll get 4104 bytes voice bank. Headers included.
4. same for the function part of the voices - single or bulk dump requests
5. the function part of the voices translate to the performance parameters on MiniDexed

Skerjanc, 2025-04-29:
> ### Test on TX816
> Set up the TX816 like this:
> 
> 1. First module MIDI channel 9, second module MIDI channel 10,... eighth module channel 16
> 2. Set the "Output Slot" button on the front panel to the first slot (module)
> 3. Set a different distinguishable voice for every module (e.g., first module "BRASS 1", second module "BRASS 2",... eigth module "PIANO 1")
> 4. Send a voice dump request for device ID 0 - what voice gets dumped (if any)? What is the 3rd byte of the dump?
> 5. Send a voice dump request for device ID 7 - what voice gets dumped (if any)? What is the 3rd byte of the dump?
> 6. Send a voice dump request for device ID 15 - what voice gets dumped (if any)? What is the 3rd byte of the dump?
> 7. Increment the "Output Slot" button on the front panel and repeat steps 4.-7.

It is rather simple:

[image]
The dump request is based on MIDI channel nnnn in 3rd byte. While all modules receive the request, only the one with the matching MIDI channel will respond. In the example above only module 8 will respond to req. 6, nothing else.
Respond either through the MIDI Out at the back side or the MIDI out in the front side, when the module number has been selected.

BobanSpasic, 2025-04-29:
Btw. from your opening post here - the sub-status defines the type of the message:
s=0 - voice/suppl./perf. dump
s=1 - direct single parameter change
s=2 - dump request

So, at requesting you use s=2, and you receive the dump with s=0.

BobanSpasic, 2025-04-29:
As for the f (or ff in your post):
    f=0 - single_voice DX7
    f=1 - single_function TX7
    f=2 - bulk_function TX7
    f=3 - single_voice DX11/TX81z/V50/DX21
    f=4 - bulk_voice DX11/TX81z/V50/DX21
    f=5 - single_supplement DX7II
    f=6 - bulk_supplement DX7II
    f=7 -
    f=8 -
    f=9 - bulk_voice DX7
    f=7E - SysExUniversalDump

The last one (Universal dump) has string-identifier for the data. As the Yamaha got to the conclusion, that there won't be enough f-numbers for all the future synths, they got to the idea of 7E with additional string. Here are some of them (DX-related, underlines are spaces and **x** are numbers):

DX7II
    LM__8973PE    61 byte   DX7II Performance Edit Buffer                     1x
    LM__8973PM  1642 byte   DX7II Packed 32 Performance                       1x
    LM__8973S_   112 byte   DX7II System Set-up                               1x
    LM__MCRYE_   266 byte   Micro Tuning Edit Buffer                          1x
    LM__MCRYMx   266 byte   Micro Tuning with Memory #x=(0,1)                 2x
    LM__MCRYC_   266 byte   Micro Tuning Cartridge                           64x
    LM__FKSYE_   502 byte   Fractional Scaling Edit Buffer                    1x
    LM__FKSYC_   502 byte   Fractional Scaling in Cartridge with Memory #    32x

V50/DX11/TX81z
    LM__8976AE    33 byte   ACED    TX81Z
    LM__8023AE    20 byte   ACED2   DX11
    LM__8073AE    30 byte   ACED3   V50
    LM__8976PE   120 byte   PCED    DX11
    LM__8073PE    43 byte   PCED2   V50
    LM__8976PM  2442 byte   PMEM    DX11
    LM__8073PM   810 byte   PMEM2   V50
    LM__8976Sx    xx byte   System
    LM__MCRTE0    34 byte   Micro Tuning Edit Buffer OCT
    LM__MCRTE1   274 byte   Micro Tuning Edit Buffer FULL
    LM__8023S0    26 byte   System
    LM__8073S0    42 byte   System

Most of the modern Yamaha keyboards use universal dump with some other string identifiers.

Skerjanc, 2025-04-29:
> So, at requesting you use s=2, and you receive the dump with s=0.

As far as I see this, yes.

Skerjanc, 2025-04-29:
> Which poses the question, what happens if more than one module (with different voices) is set to the MIDI channel that a dump request ist sent to? (For MiniDexed, I would then go with "dump the voice of the first TG that matches the MIDI channel for which the dump is.)

This will be a conflict. A TX816 always wants to have its modules set to different MIDI channels.

probonopd, 2025-04-29:
So on MiniDexed, I'll conceptually replace "edit buffer" with "the data loaded into the first TG that matches the requested MIDI channel".

PR #899 (2025-05-10): Honor voice dump requests; make rtpMIDI bi-directional
Partially resolves #882 (so far only for voice dump requests), closes #898
closed 2025-05-10 by probonopd