probonopd/MiniDexed #889: Glissando effect seems to affect all operators differently

probonopd, 2025-04-25:
@skerjanc [reported](https://github.com/probonopd/MiniDexed/pull/884#issuecomment-2828982690):

> Glissando SYSEX catches the right function, but the glissando effect seems to affect all operators differently, thus it sounds detuned in steps. An issue of Dexed also?

probonopd, 2025-04-25:
I think we need a way of testing this on MiniDexed and on real hardware for comparison.

Would you be able to run the script below with MiniDexed and module 1 of your TX816 for comparison? Do you hear any differences between the two? (I know it is a bit much to ask, but if you hear differences, could you record them for comparison?)

```
#!/usr/bin/env python
# -*- coding: utf-8 -*- 

# Test MIDI portamento and glissando features of T816

import time
import sys
import signal
import rtmidi

def ask(question, play_fn=None):
    while True:
        reply = input(f"{question} (y/n/r): ").strip().lower()
        if reply == 'r':
            if play_fn:
                play_fn()
            continue
        if reply in ['y', 'n']:
            return reply == 'y'

def send_sysex(midiout, msg, label=""):
    # Force all SysEx messages to MIDI channel 0 (0x10)
    if len(msg) > 2 and msg[0] == 0xF0 and msg[1] == 0x43:
        msg = msg.copy()
        msg[2] = 0x10
    midiout.send_message(msg)
    print(f"Sent: {label}")

def send_parameter(midiout, parameter, value, label):
    """Helper function to send SysEx for a single parameter."""
    msg = [0xF0, 0x43, 0x10, 0x04, parameter, value, 0xF7]
    send_sysex(midiout, msg, label)

def note_on(midiout, note, velocity=100, channel=0):
    midiout.send_message([0x90 | 0, note, velocity])

def note_off(midiout, note, channel=0):
    midiout.send_message([0x80 | 0, note, 0])

def play_legato(midiout, notes, delay=0.5, channel=0):
    midiout.send_message([0x90 | 0, notes[0], 100])
    time.sleep(delay)
    midiout.send_message([0x90 | 0, notes[1], 100])
    time.sleep(delay)
    midiout.send_message([0x80 | 0, notes[0], 0])
    midiout.send_message([0x80 | 0, notes[1], 0])

def play_sequential(midiout, notes, delay=0.5, channel=0):
    note_on(midiout, notes[0], channel=0)
    time.sleep(delay)
    note_off(midiout, notes[0], channel=0)
    time.sleep(0.1)
    note_on(midiout, notes[1], channel=0)
    time.sleep(delay)
    note_off(midiout, notes[1], channel=0)

def play_chord(midiout, notes, velocity=100, duration=1.0, channel=0):
    for note in notes:
        note_on(midiout, note, velocity, 0)
    time.sleep(duration)
    for note in notes:
        note_off(midiout, note, 0)

def get_port():
    midiout = rtmidi.MidiOut()
[... 175 lines cut ...]
    for name, passed, sysex in results:
        if passed:
            print(f"PASS: {name}")
        else:
            print(f"FAIL: {name}")
            if isinstance(sysex[0], list):
                for msg in sysex:
                    hexstr = ' '.join(f'{b:02X}' for b in msg)
                    print(f"  To reproduce: {hexstr}")
            else:
                hexstr = ' '.join(f'{b:02X}' for b in sysex)
                print(f"  To reproduce: {hexstr}")

if __name__ == "__main__":
    signal.signal(signal.SIGINT, lambda s, f: sys.exit(0))
    midiout = get_port()
    try:
        print("\n--- POLY MODE TESTS ---")
        test_sequence(midiout, is_mono=False)

        print("\n--- MONO MODE TESTS ---")
        test_sequence(midiout, is_mono=True)
    finally:
        del midiout
```

Skerjanc, 2025-04-27:
Sorry, I am not the python expert. I am getting errors like:
  File "D:\temp\TestGlissando.py", line 66, in get_port
    midiout = rtmidi.MidiOut()
AttributeError: module 'rtmidi' has no attribute 'MidiOut'. Did you mean: 'RtMidiOut'?

probonopd, 2025-04-28:
Oops, I think I had only tested the Python script on Linux. So let's not use it for now.

Can you come up with a sequence of MIDI sysex that you can share and could run/record on both MiniDexed and TX816 so that we hear the difference? I think that would be a tremendous help. Thanks!

Skerjanc, 2025-04-29:
You can simply hear it on MiniDexed following these steps:
1. TG1>Portamento>Time = 70 : you will hear a long portamento between two notes.
2. TG1>Portamento>Glissando On: you will hear the portamento quantized in semitones as expected, but totally detuned.

In this state the detune parameters of the operators are no longer properly working, causing crazy settings.

probonopd, 2025-04-29:
Minimal viable reproduction:

- Performance -> Bank -> Laboratory
- Performance -> Load -> 128:003 pure DX7
- TG1 -> Portamento -> Time -> 1
- Play only C note (no other notes)
- TG1 -> Portamento -> Glissando -> On
- Play only C note (no other notes) again. __The note is now deeper (detuned)!
- TG1 -> Portamento -> Time -> 0
- Play only C note (no other notes) again. __The note is now higher again (as it was originally). Changing other values for Time other than 0 detunes the note, but the amount of time does not change the amount of detuning.

probonopd, 2025-04-29:
Thanks @Skerjanc for having [confirmed](https://github.com/probonopd/MiniDexed/pull/897#issuecomment-2840174310) that this patch works.

For now, we are patching the [Synth_Dexed](https://codeberg.org/dcoredump/Synth_Dexed) engine on the fly, but a proper solution would be to get this fixed there. So maybe @smuehlst or @soyersoyer (who has sent successful Pull Requests to Synth_Dexed in the past) want to have a look at this? That'd be tremendous. Thanks!

dcoredump, 2025-05-01:
I don't have any hardware to test at the moment and hope I haven't done anything wrong. Can you please test [2ad9e43095](https://codeberg.org/dcoredump/Synth_Dexed)?

probonopd, 2025-05-01:
Tested, works. Thanks 👍

soyersoyer, 2025-07-25:
This is solved, right?

probonopd, 2025-07-25:
Yes. Thank you very much.