probonopd/MiniDexed #908: Click/Pop sound on startup

Octet-nl, 2025-05-03:
Hi,

When I start MiniDexed, or I reset the device, Right after the build number is shown and directly before TG1 shows, a click/pop sound is heard in the speakers:

[MiniDexed startup pop_click.zip](https://github.com/user-attachments/files/20023189/MiniDexed.startup.pop_click.zip)

I'm using a PCM5102A I2S soundcard. When I listen with a headphone, it's the right channel (unless I swapped them).

When I turn the master volume down to 0, there is no click/pop sound, when I turn it up higher (from 1 up) then it is always heard at the same volume, so it does not get louder when the master volume is turned higher.

Is there something that can be done about this? A kind of soft start would help I think.

Kind regards,

Hans

probonopd, 2025-05-03:
Interesting observation. So you say by setting master volume to 0, then restarting the device, and then turning master volume up again you hear no click?

So far, I thought the click was in hardware, but this would suggest otherwise.

Octet-nl, 2025-05-03:
Correct,

I just checked (first time I forgot to turn master volume up again...).

Setting master volume to 0 in minidexe.ini and turn it up after starting I hear no click.

soyersoyer, 2025-05-03:
Nice catch! There is a buffer size mismatch at start.
It's quiet If I add
```
				arm_fill_f32(0.0f, SampleBuffer[indexL], nFrames);
				arm_fill_f32(0.0f, SampleBuffer[indexR], nFrames);
```
before:
```
tg_mixer->getMix(SampleBuffer[indexL], SampleBuffer[indexR]);
```
Maybe the nFrames should be limited to pConfig->GetChunkSize()/2 because the tg_mixer cannot mix more frames.

soyersoyer, 2025-05-03:
But this is not the best solution yet :) I'll find out tonight what exactly this m_nQueueSizeFrames logic is.

This also works:
```
	if (nFrames >= m_nQueueSizeFrames/2)
	{
		nFrames = std::min(nFrames, m_pConfig->GetChunkSize()/2);
```

probonopd, 2025-05-04:
@soyersoyer would you like to send another PR to solve this? Then I'll close #910.

PR #912 (2025-05-05): limit the number of processed frames to chunksize / 2
The tg_mixer cannot process more franes,
So if it has to process more, uninitialized frames will be played, which causes strange sounds, e.g. at startup.

Closes #908
commit d70232dc (2025-05-05): Remove startup sound artefact

Limit the number of processed frames to chunksize / 2 (#912)
The tg_mixer cannot process more franes,
So if it has to process more, uninitialized frames will be played, which causes strange sounds, e.g. at startup.

Closes #908
closed 2025-05-05 by probonopd