probonopd/MiniDexed #916: Portamento Memory does not behave like on TX816

probonopd, 2025-05-12:
@Skerjanc in https://github.com/probonopd/MiniDexed/pull/884#issuecomment-2872484797:

**Portamento Memory** is implemented wrong: currently it only enables/disables portamento. 
But on the HW it does the following (**only in Monophonic Mode**):
- Off: Portamento is disabled when two notes are played non-legato. Played legato, portamento is enabled
- On: always portamento

BobanSpasic, 2025-05-12:
What you are describing is the difference between Fingered and Full time porta, not between Portamento on and off. Both available just in Mono mode when Portamento is On (and Glisando is off).
In Poly mode, you have Sus-key P Retain and Sus-key P Follow - If you use a Sustain pedal, what should happen with the sustained tone - should it remain on its own frequency, or should it slide to the next pressed key.

probonopd, 2025-05-12:
@Skerjanc, based on the feedback from @BobanSpasic, can you please compare what happens when you change from "Fingered" (default) to "Full time" under TGx -> Portamento -> Mode?

probonopd, 2025-05-14:
@Skerjanc I understand that the behavior is not correct regardless of whether you use MIDI SysEx or the menu on the DT-DX, right?

Possibly this needs to be addressed in the [Synth_Dexed](https://codeberg.org/dcoredump/Synth_Dexed) engine along these lines:

```
    if (monoMode) {
        // Determine portamento mode: 0 = Fingered, 1 = Full Time
        uint8_t portamento_mode = controllers.portamento_enable_cc ? 1 : 0;
        // Count how many notes are currently held (keydown == true)
        int numHeld = 0;
        for (uint8_t i = 0; i < used_notes; ++i) {
            if (voices[i].keydown) numHeld++;
        }
        if (portamento_mode == 0) { // Fingered: only legato
            // Only enable portamento if another note is held (legato)
            if (numHeld > 0) {
                porta = controllers.portamento_cc;
            } else {
                porta = -1;
            }
        } else { // Full Time: always portamento
            porta = controllers.portamento_cc;
        }
    } else {
        // Polyphonic mode
        if (controllers.portamento_enable_cc && previousKeyDown >= 0)
            porta = controllers.portamento_cc;
    }
```

@dcoredump what do you think?

dcoredump, 2025-05-15:
I have to get my DX7 back from studio to check and compare  this. I will try after I am back from holidays.

Perhaps someone can create a PR?

Skerjanc, 2025-05-15:
> [@Skerjanc](https://github.com/Skerjanc) I understand that the behavior is not correct regardless of whether you use MIDI SysEx or the menu on the DT-DX, right?

Yes. 
Please don't get me wrong, when I say the portamento time is reset to 0. I mean the **parameter** Portamento Time becomes 0, (thus the portamento effect vanishes) when setting fingered mode.