probonopd/MiniDexed #930: Continuous build "MiniDexed_1181_2025-05-10-6a9d3a6.zip" unstable

Banana71, 2025-06-08:
With my Raspberry Pi 4, I may experience crackling in the sound output after receiving Sysex data.

I encounter this problem when, for example, I edit all 8 TGs simultaneously on the same MIDI channel.

https://github.com/user-attachments/assets/a3ac4c28-4969-4e6d-ac44-11382cf358c0

In the video I loaded the performance 001:041 Barock and then transferred the INIT patch to all 8 TGs and changed parameters that shouldn't actually have any influence on the sound.

probonopd, 2025-06-08:
Hi @Banana71, thanks for reporting. 

Does the crackling persist or is it only while (shortly after) the knob is being turned?

Banana71, 2025-06-08:
Once it cracks, it stays that way. Only after further changes to the values does it improve. But not every value change produces this effect.

probonopd, 2025-06-09:
Is this happening over all MIDI inputs (serial, USB, network) or just some?

Banana71, 2025-06-09:
Crackling also occurs when connecting via rtbMIDI (RPi 4)

By the way, I'm running the version "MiniDexed_1142_2025-04-29-1199230" without any problems. Unfortunately, I have no idea what was implemented after that.

Banana71, 2025-06-09:
I've tested the following versions, but the most recent one seems to be causing the problems.

MiniDexed_1142_2025-04-29-1199230 - OK
MiniDexed_1174_2025-05-04-e127f61_64bit - OK
MiniDexed_1176_2025-05-10-112681d_64bit - OK
MiniDexed_1180_2025-05-10-c21751e_64bit - Crackling https://github.com/probonopd/MiniDexed/pull/899#issuecomment-2869124103