probonopd/MiniDexed #956: Network connection not working in Continuous build

Banana71, 2025-07-12:
I haven't tried every continuous build lately, but I've just noticed that the network connection on my Raspberry Pi 4 is no longer working. 
- The IP address is no longer displayed during boot. 
- The "Firmware" folder is no longer included in the ZIP file, and the "wpa_supplicant.conf" file is also missing. (Both manually copied to the SD card.)

The miniDexed is displayed in the router.
FTP connection cannot be established.
DHCP on or off, both unsuccessful.

Peter 

[image]

probonopd, 2025-07-12:
Thanks for reporting. Can you identify which was the last build that worked?

Banana71, 2025-07-13:
[image]

Starting from this page [commints - Main](https://github.com/probonopd/MiniDexed/commits/main/)

- #899 MiniDexed_1180_2025-05-10-c21751e_64bit.zip, the network works perfectly.
- #944 MiniDexed_1194_2025-07-06-9ebb5f8_64bit.zip: no network. From this point on, the folder structure was also unusual.

An error probably crept in sometime after May 10, 2025.

Banana71, 2025-07-13:
> What happens if you take the version from MiniDexed_1180_2025-05-10-c21751e_64bit.zip (make sure that networking works) and then ONLY update the kernel* files with the one from the latest continuous build?

Yes, that's exactly what I did. In the version from May 10th, the network works.

Peter

probonopd, 2025-07-20:
@Banana71 if you have a chance, please test with the latest continuous build.

Banana71, 2025-07-23:
> [@Banana71](https://github.com/Banana71) if you have a chance, please test with the latest continuous build. Thanks!

Unfortunately, this did not bring any improvement, the network still only works for me up to the version from May 10th

Banana71, 2025-07-23:
> It shouldn't work with the May 10 version with the new kernel, or the latest should work.

I don't understand what you're saying 🤷‍♂️

soyersoyer, 2025-07-23:
> > What happens if you take the version from MiniDexed_1180_2025-05-10-c21751e_64bit.zip (make sure that networking works) and then ONLY update the kernel* files with the one from the latest continuous build?
> 
> Yes, that's exactly what I did. In the version from May 10th, the network works.
> 
> Peter

Here, you updated only the kernel files and it worked.
The new version is exactly the same as the May 10th version, except for the kernel* and performance files.

probonopd, 2025-07-23:
@soyersoyer could it be one of the code changes made after May 10 to main?

Banana71, 2025-07-23:
https://github.com/probonopd/MiniDexed/pull/962#issuecomment-3109965700 
solved the problem, many thx @soyersoyer

PR #962 (2025-07-23): submod: update circle/addon/wlan/hostap
> The issue should be fixed on the _develop_ branch. This required a modification on the _addon/wlan/hostap_ submodule too.

https://github.com/rsta2/circle/issues/586#issuecomment-2966223109

probably fixes #956
closed 2025-07-23 by Banana71