probonopd/MiniDexed #967: MIDIThru=umidi1,udp Network Midi Thru isn't making sound

RK11111111111, 2025-07-27:
If I set MIDIThru using rtpmidi, MiniDexed network sends midi and I can see it on a remote computer, however when I do this it does not make a sound.  I can disable this rtpMIDI connects but doesn't send the midi from the keyboard.  It just sends the midi.  Even if the network is not connected, the synth doesn't make a sound. 

MIDIThru=umidi1,udp 

Just for completeness.  I want to be able to record the midi I play at the keyboard live while it is making sound, then play it back with a midi sequencer.

probonopd, 2025-07-28:
If I am not mistaken, MIDI Thru is designed to take something from a MIDI input and put it to a MIDI output.

So you need to tell it to forward the MIDI data from your keyboard to your computer.

How is your keyboard attached to MiniDexed?

probonopd, 2025-07-29:
> MiniDexed should function if the network fails. 

It absolutely should; if it doesn't, it's a bug.

RK11111111111, 2025-07-30:
Should be able to isolate a little.  I mean if the network isn't there on boot.  Do I have to swap in a different sd card to play?  With my other network enable synths and sound toys, I run a script that checks if it is connected every about 8 seconds, if it isn't and they are available devices it connects.    It's not horribly complicated to do.

probonopd, 2025-07-30:
If the network is not available on boot, then MiniDexed should work exactly as if no network was there to begin with. If this is not the case, please open a bug with details on how to reproduce. Thanks

RK11111111111, 2025-07-30:
This is essentially what this issue is.  Unplug ethernet cable.  MiniDexed boots but no sound.  Plug the cable into an active ethernet restart and it works.