probonopd/MiniDexed #976: Build from source for pi-zero-2 on macos

marchingband, 2025-08-03:
If I use your release .zip file, my pi-zero-2 works well.
However I am struggling to build from source for this target.
I set `RPI=3`, and followed the steps at https://github.com/probonopd/MiniDexed/wiki/Development#building-locally, installing https://developer.arm.com/-/media/Files/downloads/gnu-a/10.3-2021.07/binrel/gcc-arm-10.3-2021.07-x86_64-aarch64-none-elf.tar.xz.

`./build.sh` fails with
```
+ '[' -z 3 ']'
+ '[' 3 -gt 2 ']'
+ export TOOLCHAIN_PREFIX=aarch64-none-elf-
+ TOOLCHAIN_PREFIX=aarch64-none-elf-
+ OPTIONS='-o USE_PWM_AUDIO_ON_ZERO -o SAVE_VFP_REGS_ON_IRQ -o REALTIME -o SCREEN_DMA_BURST_LENGTH=1'
+ '[' 3 -gt 1 ']'
+ OPTIONS='-o USE_PWM_AUDIO_ON_ZERO -o SAVE_VFP_REGS_ON_IRQ -o REALTIME -o SCREEN_DMA_BURST_LENGTH=1 -o ARM_ALLOW_MULTI_CORE'
+ '[' 3 == 3 ']'
+ OPTIONS='-o USE_PWM_AUDIO_ON_ZERO -o SAVE_VFP_REGS_ON_IRQ -o REALTIME -o SCREEN_DMA_BURST_LENGTH=1 -o ARM_ALLOW_MULTI_CORE -o USE_SDHOST'
+ source USBID.sh
++ USB_VID=0x1209
++ USB_DID=0xF043
+ '[' 0x1209 ']'
+ OPTIONS='-o USE_PWM_AUDIO_ON_ZERO -o SAVE_VFP_REGS_ON_IRQ -o REALTIME -o SCREEN_DMA_BURST_LENGTH=1 -o ARM_ALLOW_MULTI_CORE -o USE_SDHOST -o USB_GADGET_VENDOR_ID=0x1209'
+ '[' 0xF043 ']'
+ OPTIONS='-o USE_PWM_AUDIO_ON_ZERO -o SAVE_VFP_REGS_ON_IRQ -o REALTIME -o SCREEN_DMA_BURST_LENGTH=1 -o ARM_ALLOW_MULTI_CORE -o USE_SDHOST -o USB_GADGET_VENDOR_ID=0x1209 -o USB_GADGET_DEVICE_ID_BASE=0xF043'
+ cd circle-stdlib/
+ make mrproper
Makefile:3: Config.mk: No such file or directory
make: *** No rule to make target `Config.mk'.  Stop.
+ true
+ ./configure -r 3 --prefix aarch64-none-elf- -o USE_PWM_AUDIO_ON_ZERO -o SAVE_VFP_REGS_ON_IRQ -o REALTIME -o SCREEN_DMA_BURST_LENGTH=1 -o ARM_ALLOW_MULTI_CORE -o USE_SDHOST -o USB_GADGET_VENDOR_ID=0x1209 -o USB_GADGET_DEVICE_ID_BASE=0xF043 -o KERNEL_MAX_SIZE=0x400000
RASPBERRYPI=3
TOOLPREFIX=aarch64-none-elf-
CIRCLE_BUILD=1
CIRCLE_QEMU=0
FLOAT_ABI=hard
NEWLIB_BUILD=1
RASPBERRYPI=3
STDLIB_SUPPORT=3
OPTIONS= -DUSE_PWM_AUDIO_ON_ZERO -DSAVE_VFP_REGS_ON_IRQ -DREALTIME -DSCREEN_DMA_BURST_LENGTH=1 -DARM_ALLOW_MULTI_CORE -DUSE_SDHOST -DUSB_GADGET_VENDOR_ID=0x1209 -DUSB_GADGET_DEVICE_ID_BASE=0xF043 -DKERNEL_MAX_SIZE=0x400000
NEWLIB_INSTALL_DIR=/Users/andy/MiniDexed/circle-stdlib/install
NEWLIB_BUILD_DIR=/Users/andy/MiniDexed/circle-stdlib/build/circle-newlib
DEBUG=0
Remaining arguments:
Error: Unable to determine include path for stddef.h, exiting
STDDEF_INCPATH is ""
```
I have been using Circle successfully for while, but this is my first time adding circle-stdlib.
I am on macos.
Any tips, anything I am missing?

Would it make sense to fork, and use github actions to build?

thank you.

marchingband, 2025-08-03:
Interesting to note that I can build it fine in 32bit mode with `RPI=1` and run it on a pi-zero-1, but I cannot run this on a pi-zero-2, even if I edit `config.txt`

```
# Zero 2 W
[pi02]
arm_64bit=0
kernel=kernel.img
```

marchingband, 2025-08-03:
I also installed I installed arm-gnu-toolchain-14.3.rel1-darwin-arm64-aarch64-none-elf
but get the error `TOOLPREFIX is "aarch64-none-elf-"`

probonopd, 2025-08-03:
Hello @marchingband 

> Would it make sense to fork, and use github actions to build?

Yes, then it should build.

Probably the key is running `submod.sh` as described in https://github.com/probonopd/MiniDexed/wiki/Development

marchingband, 2025-08-03:
> Probably the key is running `submod.sh` as described in https://github.com/probonopd/MiniDexed/wiki/Development

I followed each of the steps carefully.
I believe the issue is that a gcc 10.3 toolchain does not exist for mac silicon, and using 13.3 does not have the same header conventions.
It's strange because I can build MiniDexed for 32 bit target with 13.3, only the 64 bit fails.

I may try to install docker, as the easiest way, if you don't have any more suggestions.

soyersoyer, 2025-08-04:
Hey marchingband, check out #977 , that should work.

marchingband, 2025-08-04:
@soyersoyer heyyyy that worked! thank you so very much! Mac users rejoice!